meshtastic/firmware #224: sleep/wake ttgo-lora32-v1 fails at "Starting bluetooth" after lsEnter end

aHVzY2g, 2020-06-24:
In the current master branch, the 'ttgo-lora32-v1' fails at "Starting Bluetooth" after "lsEnter end" also it doesn't react to a button press. The Issue doesn't exist at version 0.7.8 . I checked the commits science 0.7.8 but couldn't find something that could cause this issue.
Can someone confirm the issue with another board?

Observations:
- last serial output "Starting Bluetooth"
- doesn't react to a button press
- LED is continuously on(GPIO2 HIGH)

Here is the Serial output:
_Current-Master_
```
Screen: Started...
Starting low level send (id=0x730c2aa9 Fr0x7c To0xff, WantAck0, HopLim3 encrypted)
showing standard frames
Transition powerFSM transition=boot timeout, from=BOOT to=ON
Setting bluetooth enable=1
Pre BT: 208400 heap size
Starting bluetooth
*** Mesh service:
Completed sending (id=0x730c2aa9 Fr0x7c To0xff, WantAck0, HopLim3 encrypted)
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
Turning off screen
Trigger powerFSM 1
Transition powerFSM transition=Press, from=DARK to=ON
Turning on screen
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
Turning off screen

Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
lsEnter begin, ls_secs=3600
lsEnter end
Setting bluetooth enable=0
Shutdown BT: 206644 heap size
wakeCause 4
Trigger powerFSM 2
Transition powerFSM transition=Wake timer, from=LS to=DARK
Setting bluetooth enable=1
Pre BT: 206644 heap size
Starting bluetooth
```

_0.7.8_
```
Screen: Started...
Starting low level send (id=0x0b4829b2 Fr0x7c To0xff, WantAck0, HopLim3 encrypted)
showing standard frames
Transition powerFSM transition=boot timeout, from=BOOT to=ON
Setting bluetooth enable=1
Pre BT: 214112 heap size
Starting bluetooth
*** Mesh service:
Completed sending (id=0x0b4829b2 Fr0x7c To0xff, WantAck0, HopLim3 encrypted)
Transition powerFSM transition=Screen-on timeout, from=ON to=DARK
Turning off screen

Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
lsEnter begin, ls_secs=3600
lsEnter end
Setting bluetooth enable=0
Shutdown BT: 212344 heap size
wakeCause 4
Trigger powerFSM 2
Transition powerFSM transition=Wake timer, from=LS to=DARK
Setting bluetooth enable=1
Pre BT: 212344 heap size
Starting bluetooth

*** Mesh service:
Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
lsEnter begin, ls_secs=3600
lsEnter end
Setting bluetooth enable=0
Shutdown BT: 212344 heap size
wakeCause 4
Trigger powerFSM 2
Transition powerFSM transition=Wake timer, from=LS to=DARK
Setting bluetooth enable=1
Pre BT: 212344 heap size
Starting bluetooth
*** Mesh service:
Transition powerFSM transition=Bluetooth timeout, from=DARK to=LS
lsEnter begin, ls_secs=3600
lsEnter end
Setting bluetooth enable=0
Shutdown BT: 212344 heap size
Exit light sleep gpio: btn=1
wakeCause 7
Trigger powerFSM 1
Transition powerFSM transition=Press, from=LS to=ON
Setting bluetooth enable=1
Pre BT: 212344 heap size
Starting bluetooth
*** Mesh service:
Turning on screen
```

aHVzY2g, 2020-06-24:
Just found "check BLE handle stability across sleep - stress test sleep/wake - btu_init_core calls gatt_init - which assigns handles global" in the TODO. I guess that is related to this issue 🤷

geeksville, 2020-06-25:
@aHVzY2g hmm - it shouldn't be.  Can you try the following to confirm you got my fixed arduino-esp32?  In the project directory run "platformio lib update" and see if it pulls down anything new?

aHVzY2g, 2020-06-25:
@geeksville That will be the Issue, the # in the URL is considered as a comment by the compiler :D

geeksville, 2020-06-25:
ooooh!  interesting!

aHVzY2g, 2020-06-25:
Nevermind it's not considered as a comment. Too bad, that would have been an easy fix. 
So yes it compiles correctly, I can see it during compiling "...#f26c4f96fefd13ed0ed042e27954f8aba6328f6b"

aHVzY2g, 2020-06-25:
after doing a deep clean of all dependencies, the issue seems to be resolved. Thx for the pointer to the framework

closed 2020-06-25 by aHVzY2g