meshtastic/firmware #235: GATT features request

slavino, 2020-06-27:
# Standard GATT

### DateTime from GPS
https://www.bluetooth.com/xml-viewer/?src=https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.date_time.xml

### Time source - GPS
https://www.bluetooth.com/xml-viewer/?src=https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.time_source.xml

### Longitude
https://www.bluetooth.com/xml-viewer/?src=https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.longitude.xml

### Latitude
https://www.bluetooth.com/xml-viewer/?src=https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.latitude.xml

### Altitude
https://www.bluetooth.com/xml-viewer/?src=https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.altitude.xml

### and maybe freestyle other environment data if sensor present or received from mesh (air pressure, ...)
http://arduiniana.org/libraries/tinygpsplus/

geeksville, 2020-06-28:
good idea and fairly easy to do.

qurm, 2020-10-16:
Hi,

I am interested in developing this enhancement.  I have done some hobby BLE work on NRF devices a year or so back.  I have been looking around for reference implementations, but there's not much out there, and very little on NimBLE.  I have the LNS spec from here, and can probably work from that, with NimBLE examples for other services:  https://www.bluetooth.com/specifications/gatt/

Also there seem to be some BLE apps that provide GPS data over BT, but it is not clear that they use the GATT LNS service. I suspect that they stream NMEA data some other way (UART?). Does anyone know more about this?  like https://play.google.com/store/apps/details?id=com.Saenko.GpsOverBt  
@slavino -do you have an application that you would use this with?

closed 2022-02-02 by garthvh