meshtastic/firmware #242: Screen doesn't change to show received message.

Dafeman, 2020-07-02:
Once paired to the TBeam, upon sending a message from the connected phone, it used to display the message on the OLED of the TBeam once received (almost instant). On 0.7.10 firmware it can be sporadic. Sometimes the message is displayed but only if you scroll to the that screen with the center button, sometimes it isn't displayed at all. On all occasions the message was still sent and received by the other TBeam. Likewise however, the receiving TBeam didn't show the message unless already on the message screen.

**Device info:**
 - Device model: TBEAM
 - Software Version: 0.7.10

**Smartphone information (if relevant):**
 - Device: ZUK Z2 Pro
 - OS: Android 8.0.0
 - App Version: 0.7.84

geeksville, 2020-07-03:
yes - I think this broke with the recent display apparence improvements?  Will look into it soonish.

Professr, 2020-07-03:
I can't quite figure out how the messages get there in the first place.  Nothing seems to be setting the logBuffer directly, and I don't see any single point in the MeshService that looks like it would drive that, but it does get complicated in there.
Even with the 1fps refresh when idling, it should redraw the frames often enough for the logBuffer contents to get thrown onto the screen.  If there's an observer I can have screen.cpp subscribe to and we can figure out what needs to be triggered to get the message to update, I can certainly add that in.

Dafeman, 2020-07-03:
I really don't know what I'm looking for but could it be related to this deletion in screen.cpp?

[image]

Professr, 2020-07-03:
Good find!  NodeStatus was only notifying observers if the number of nodes (online or total) changed, but there are some incoming packets that would trigger the code in that deletion even though the number of nodes stays the same.  I've added code to allow for forcing that notification to go out regardless of node count in those situations.  Can you give [this branch](https://github.com/Professr/Meshtastic-device/tree/issue%23242) a try and see if it works better?  If so, I'll make a PR

Professr, 2020-07-03:
Well, it does its job, but it highlights another issue with `setFrames()` lol.  Back to the code I go!

Professr, 2020-07-03:
`setFrames()` doesn't seem to keep track of the frame we're currently on and restore it when the frame list gets rebuilt.
- If we're on a nodeInfo frame, we should store that node num and set the frame to its index once the frame list is rebuilt
- If we're on the stats screen, we should set the frame to the stats screen index after the frame list is rebuilt
- This all goes out the window if we're `IN_TRANSITION` because the user is trying to switch frames with the button

this hurts my brain

Professr, 2020-07-04:
[latest branch commit](https://github.com/Professr/Meshtastic-device/tree/issue%23242)
I don't think `setFrames()` is intended to be called for every node update.  None of the underlying issues from my last comment have been fixed, but I changed the `handleStatusUpdate()` function to only call `setFrames()` when a text message arrives or the number of discovered nodes changes.  It seems to work better.  I wish we had a Slack or something, then maybe someone could help me understand how the ui frames are supposed to be used :\

Dafeman, 2020-07-04:
I wish I had something juicy to help with the brain hurt! But nothing other than I tried your build and got the behavior that I expected. Screen changed to display the message on receiving a message over Bluetooth and LoRa. Nice work! 

Sorry I don't know the best method for discussion. I presume the forum has the most eyes cast over it though and would be the best repository of info for future searches.

geeksville, 2020-07-04:
@Professr re something like a slack
good idea - I'll look into reviving the original gitter channel (used before we switch to discourse) as a "developer chat"

geeksville, 2020-07-04:
actually - slack might be even better than gitter because of this https://meta.discourse.org/t/chatroom-integration-plugin-discourse-chat-integration/66522

geeksville, 2020-07-04:
invite link here: https://join.slack.com/t/meshtasticdev/shared_invite/zt-fwo0ai9c-5ywd5nx3klPyL_LzYusy5Q

Professr, 2020-07-05:
It works pretty well in my tests too.  We'll want do something more structured later, but this should take care of the immediate problem

PR #256 (2020-07-06): Issue#242 - at least partially fixed the setFrames issue with nodeDB updates
I added the ability for NodeDB to notify observers of events that should force propagate even if the number of online / total nodes did not change.  I integrated that into screen.cpp to fix the issue with incoming messages not being displayed, and some problems with being able to switch between frames with the user button (partially fixes #242).  I also tweaked the compass display to better handle an edge case
closed 2020-07-06 by Professr