meshtastic/firmware #243: TTGO TBeam EU 433 - releases not working 0.7.9 , 0.7.10

slavino, 2020-07-02:
TTGO TBeam EU 433

Board version: T22_V07 20180711

GPS : ublox NEO-6M

-----------------------------------------

```
Using 'COM14' as serial port.
Showing logs:
[13:17:09]ets Jun  8 2016 00:22:57
[13:17:09]
[13:17:09]rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[13:17:09]configsip: 0, SPIWP:0xee
[13:17:09]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:17:09]mode:DIO, clock div:2
[13:17:09]load:0x3fff0018,len:4
[13:17:09]load:0x3fff001c,len:1044
[13:17:09]load:0x40078000,len:8896
[13:17:09]load:0x40080400,len:5828
[13:17:09]entry 0x400806ac
[13:17:13]Wc_F%g֌BjCbOeCO;#/#"i"ΞtgƔkupsJgÍی[13:17:13]
[13:17:13]rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[13:17:13]configsip: 0, SPIWP:0xee
[13:17:13]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:17:13]mode:DIO, clock div:2
[13:17:13]load:0x3fff0018,len:4
[13:17:13]load:0x3fff001c,len:1044
[13:17:13]load:0x40078000,len:8896
[13:17:13]load:0x40080400,len:5828
[13:17:13]entry 0x400806ac
[13:17:16]v_#k	fF8bNeCOj;!#Bfk~Bzug!u%eprgQF%c#C;J0z뿯魽=鯩?{8GFBcPVqfK!NBB1b!GCB9BNFC$BB	NCBBBDB@BJCCBBBBCCqNFCDC[13:17:16]
[13:17:16]rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[13:17:16]configsip: 0, SPIWP:0xee
[13:17:16]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:17:16]mode:DIO, clock div:2
[13:17:16]load:0x3fff0018,len:4
[13:17:16]load:0x3fff001c,len:1044
[13:17:16]load:0x40078000,len:8896
[13:17:16]load:0x40080400,len:5828
[13:17:16]entry 0x400806ac
[13:17:18]vc_#j%\gkFp8BbNeBN
[13:17:20];#/#Bfk?~{(Bz!+p;gDJJ#;c"B80뭵響-z￩jk魩_/#sqs!pg1-NFJBBBJDC@CJ!FBCBBBBBFB!NBCBBFJ!0BBCBFBN BʩVFBB)JPCaJBBKRZ)mIBRJIFKets Jun  8 2016 00:22:57
[13:17:20]
[13:17:20]rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[13:17:20]configsip: 0, SPIWP:0xee
[13:17:20]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:17:20]mode:DIO, clock div:2
[13:17:20]load:0x3fff0018,len:4
[13:17:20]load:0x3fff001c,len:1044
[13:17:20]load:0x40078000,len:8896
[13:17:20]load:0x40080400,len:5828
[13:17:20]entry 0x400806ac
[13:17:23]ͩ_F	FmaJkDFOk:%#Cy(vuGץ!uep;gkGcq	0)y?=ﯩ+)驭߽{BFBs!qf$CBBFBC0KDC@BJ!GBB	BCBBBFC!NBBBCFJK1CBBBFB BʩVFBB)JPCaBJ9Q-RB7BCJ䡵IZets Jun  8 2016 00:22:57
[13:17:23]
[13:17:23]rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[13:17:23]configsip: 0, SPIWP:0xee
[13:17:23]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:17:23]mode:DIO, clock div:2
[13:17:23]load:0x3fff0018,len:4
[13:17:23]load:0x3fff001c,len:1044
[13:17:23]load:0x40078000,len:8896
[13:17:23]load:0x40080400,len:5828
[13:17:23]entry 0x400806ac
[13:17:25]vc_F%^gFm{F Z$
[13:17:27]j:%ťB8J (kku*psgQFF1bsq	`鹹ﯩ??Ϲm߽.i9i('!bB+BBBK@C' CBB!B@BJJBNBBBBDB@CJBbBBBBBqCB[13:17:27]
[13:17:27]rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[13:17:27]configsip: 0, SPIWP:0xee
[13:17:27]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[... 184 lines cut ...]
[13:18:26]configsip: 0, SPIWP:0xee
[13:18:26]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:18:26]mode:DIO, clock div:2
[13:18:26]load:0x3fff0018,len:4
[13:18:26]load:0x3fff001c,len:1044
[13:18:26]load:0x40078000,len:8896
[13:18:26]load:0x40080400,len:5828
[13:18:26]entry 0x400806ac
[13:18:28]vb_N%(gkG8{bOeBO
[13:18:29]k;	#Cmo	{PCz_o+B&Ȇg(#;cCr	0뭽?//{/-y/߽/{BGBb(qg1-NGJ!G1BCBBBNFB$CCNBBBBDB@BJBBGBBCCFBN CʩVFCC[13:18:29]
[13:18:29]rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[13:18:29]configsip: 0, SPIWP:0xee
[13:18:29]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:18:29]mode:DIO, clock div:2
[13:18:29]load:0x3fff0018,len:4
[13:18:29]load:0x3fff001c,len:1044
[13:18:29]load:0x40078000,len:8896
[13:18:29]load:0x40080400,len:5828
[13:18:29]entry 0x400806ac
[13:18:29]
Serial port closed!

```

Last time after flasind 0.7.9 I had to open Arduino IDE and flash some base sketch to make it work and had to rollback to some older release (I think 0.7.8 was last non-crashing for this board).

slavino, 2020-07-02:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

With 
* TTGO Lora32 v2.1 433/470MHz (T3_V1.6 20180606) on meshtastic 0.7.10 for EU433 paired to Samsung Note 9
* TTGO T-Beam w ublox NEO-6M (T22_V07 20180711) on mestastic 0.7.8 for EU433 paired to Samsung S7
* TTGO T-Beam w ublox NEO-6M (T22_V07 20180711) on mestastic 0.7.8 for EU433 paired to Samsung S5

the position sharing works just not sure if the GPS is obtained from T-Beams. Gut feeling is that paired phone shared own location (form Android sensors GPS/AGPS/WiFi or whatever) and ublox NEO-6M is not working as it should.

slavino, 2020-07-02:
https://ae01.alicdn.com/kf/Haee084d6c33b4885b81ad176782984dfV.jpg

v0.7
[image]

vs. 

v1.1
[image]

slavino, 2020-07-02:
[image]
[image]

slavino, 2020-07-02:
I have tried to enable the platformio.ini part

[image]

but to get rid of AXP202X power management requires to dive bit deeper into the code not to break any other part of the code.

However to get the battery state I believe I used the approach from https://github.com/tekk/TTGO-T-Beam-Car-Tracker/blob/master/TTGO-T-Beam-Car-Tracker.ino 

```

#define BATTERY_PIN 35 // battery level measurement pin, here is the voltage divider connected

float vBat; // battery voltage

void getBatteryVoltage() {
  // we've set 10-bit ADC resolution 2^10=1024 and voltage divider makes it half of maximum readable value (which is 3.3V)
  vBat = analogRead(BATTERY_PIN) * 2.0 * (3.3 / 1024.0);
  Serial.print("Battery voltage: ");
  Serial.print(vBat);
  Serial.println("V");  
}

```

geeksville, 2020-07-02:
hmm - can you try capturing a serial log at a baud rate of 921600?  (It looks like you are using 115200, which is not correct)  That should make it possible to see the message.

geeksville, 2020-07-02:
@slavino  if you want to check in the change to platformio.ini that would  be great.  I only disabled it until someone could make it work, which you just did ;-)

slavino, 2020-07-03:
@geeksville ok, so first finding 

* GPS on the board is printed misleading  - it's exactly the other way round:

```
#define GPS_RX_PIN 12
#define GPS_TX_PIN 15
```
I had to check that with code that worked before on the same ver. 0.7 boards and can confirm now.

The following setting worked fine for build and upload procedure (beforehand needed to modify the configuration.h).

[image]

and of course lack of platformio knowledge made me use this "workaround" to save some time
[image]
[build.log](https://github.com/meshtastic/Meshtastic-device/files/4867919/build.log)
[runtimeLogFromPuttySerialCOM10.log](https://github.com/meshtastic/Meshtastic-device/files/4867920/runtimeLogFromPuttySerialCOM10.log)
[runtimeLogFromPuttySerialCOM14.log](https://github.com/meshtastic/Meshtastic-device/files/4867921/runtimeLogFromPuttySerialCOM14.log)
[upload.log](https://github.com/meshtastic/Meshtastic-device/files/4867922/upload.log)

Professr, 2020-07-05:
Does v07 work as expected now, after merging in these PRs?

slavino, 2020-07-06:
For me those worked just fine with local build. I'd give a chance to get a production build from GITHUB asap and start testing. 

I do also have a spare GPS module that I could attach to TTOGO Lora32 v2.1, by that my mesh network would consist of 2x T-Beam-0.7 and 1xLora32-v2.1 all equipped with GPS.

geeksville, 2020-07-06:
I think we should list TBEAM 0.7 as supported (if you build your own rom) in the readme, but given the small number of units out there I'm kinda inclined to not add it as a prebuilt bin for now.  ... remaining comment moved to a new bug:

slavino, 2020-07-07:
Well, the T-Beam v0.7 device is still being sold so I'd suggest to go for a poll showing global interest on support of boards. Or do you gather that feedback via some analytic feedback from Android App? 

(Anonymous data about Android OS version vs. Board type might be an interesting graph. What do you think? Knowing your "customer" needs might move or adjust the future focus.) Polls on feature requests might be an interesting agenda as well :)

geeksville, 2020-07-07:
ooh - I didn't realize it was still being sold!  In that case we should probably add it back to bin/build-all.sh.

We already report radio model to the analytics.  Which is a good point.  I'll look right now and do a post so people can know approximate usage!

kkwestt, 2020-07-09:
please share your .bin file for tbeam 07v

slavino, 2020-07-10:
> please share your .bin file for tbeam 07v
I believe that waiting for "production level release" is a better option than giving try to any bleeding edge build. I hope @geeksville or anybody else performing the next FW release will include this into the release itself.

PS: If the only frustration point/blocker to build your own FW is Platformio then there is some set of scripts that will execute build and flash your device via USB :)

geeksville, 2020-07-10:
done - it will be in the next build - in a couple of days.  thanks @slavino

closed 2020-07-10 by geeksville