meshtastic/firmware #252: Request for the introduction of a new function - Select radio frequency via GPS

frankenstein91, 2020-07-04:
I would like to use the function of an Internet independent messaging service while on vacation. But I don't want to have to install a new firmware every time, so I think it would be cool to introduce an automatism. But as I understand it, the hardware is bound to a country, nevertheless I try to find a solution here.

geeksville, 2020-07-04:
I agree.  At the very least I think we should stop putting differnet builds out for different frequency ranges.  It should be one build and you should be able to change frequency region from the device or the phone.

Though keep in mind: the 400ish mhz radios are physically different parts than the 900ish mhz radios.  So just changing the option won't work in that case.

frankenstein91, 2020-07-11:
sounds good to me

android606, 2021-01-28:
I'm looking at this one right now.
I'm thinking I can define a polygons with a list of GPS coordinates for each region. Then, with a little geometry, determine if the user's current GPS position is inside one of those polygons.

I'm not quite sure how to define the regions, but I have some ideas. The ITU specifies three regulatory regions:
[image]
(wikipedia)

The code currently has seven regions listed, but I think this map gives more specific information:
[image]
(http://iotfactory.eu/lorawan-world-frequency-map-lora-sigfox-nbiot-telecom-operators-per-country/)

It really looks like seven regions is not quite sufficient. I think I'm going to have to specify more regions. Maybe the list we're using is outdated?

So, I could define some large regions and some small country-sized regions. I thought about using the ITU regions as the "fallback", but Russia/Kazakhstan/China sort of make that not work. I think the "Americas" region 2 is fine, but Regions 1 and 3 won't work as-is.

Then, when checking the user's location, check against the regions in order and stop if there's a match. We check the small regions first. If there is no match, I fall back on the larger ITU-ish world-size regions.

I was thinking of mapping regions like this:

// Americas
RegionCode_FG -> French Guiana (EU868 / Same rules as France) 
RegionCode_US -> ITU Region 2 (US915 / Americas)
// Europe / Africa
RegionCode_EU433 -> Sweden (EU433)
RegionCode_KZ -> Kazakhstan (EU433)
RegionCode_AL -> Algeria (870-876, 915-921, 925-926) // NOT SURE which frequency range to use!
RegionCode_EU868 -> ITU Region 1 Plus Russia / Croatia / Kazakhstan
// Asia Pacific
RegionCode_CN -> (470-510, 779-787)
RegionCode_JP -> (920-928)
RegionCode_MY -> Malaysia (US915)
RegionCode_ANZ -> (923-925)
RegionCode_KR -> (920-923)
RegionCode_TW -> (920-925)
RegionCode_ASPAC925 -> ITU Region 3 minus Russia / Croatia / Kazakhstan

Also, I noticed that some countries in these regions have multiple frequency ranges. I only separately called out the ones that don't overlap with the greater region they're in. We may want to put more effort into specifying those regions as people demand it.

Thoughts?

slavino, 2021-01-29:
- Europe
  Slovakia is legal with both - EU868 and EU433

Only LoRaWAN providers built their network on top of 868MHz, but as said 433Mhz is legal choice as well.

Such exceptions will be hard to cover as those relate to changes in legislation.

android606, 2021-01-29:
As far I can tell, there's no way to programmatically determine the frequency range a particular board supports, either. How annoying!

Us hacker-y type people might be okay with selecting a frequency to use, but less technical people will never be able to use this if it can't enforce or at least suggest radio settings appropriate for the user's location AND compatible with their hardware.

Maybe we could at least ask the user, as part of the FW setup process, to select the frequency range listed on the sticker on their device? At least then we wouldn't be trying to offer them options that physically won't work.

closed 2021-12-12 by sachaw