meshtastic/firmware #257: Throttle button-initiated position broadcasts

Professr, 2020-07-06:
**Describe the bug**
Every button press queues a position broadcast, which gets out of hand quickly during UI navigation

**To Reproduce**
Scroll through the UI frames a few times and watch the debug console

**Expected behavior**
A button press should only send a position update if one hasn't been sent within a certain timeframe

**Device info:**
 - T-Beam T22_V1.1

geeksville, 2020-07-06:
hmm - There was  a check for "has the user seen this screen in the last two minutes?  if so, don't ask for position, if not then do"

But perhaps something is busted with that?  I haven't looked recently.

Professr, 2020-07-06:
Looks like that part is fine (though it's 30 seconds now).  The problem's in `drawNodeInfo`; every time we click past a node status frame it sends out another ping, whether we've pinged the next node recently or not.  I'll fix that, is 2 minutes too long?

geeksville, 2020-07-06:
Imo since people are mostly walking two minutes is fine.

(Sent from a phone - please ignore typos)

PR #258 (2020-07-06): Added sinceLastSeen check to pings generated by node UI
Fixes #257
closed 2020-07-06 by geeksville