meshtastic/firmware #264: Brightness slider crashes in 0.7.10 - stack trace report from forum

geeksville, 2020-07-08:
```
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ ./bin/exception_decoder.py -e ./release/archive/firmware-tbeam-US-0.7.10.elf ~/tmp/ex 
stack:
0x4017353b: OLEDDisplay::drawHorizontalLine(short, short, short) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tbeam/ESP8266_SSD1306/src/OLEDDisplay.cpp:621
0x400dccb9: OLEDDisplay::drawRect(short, short, short, short) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tbeam/ESP8266_SSD1306/src/OLEDDisplay.cpp:621
0x400dbd17: meshtastic::Screen::adjustBrightness() at /home/kevinh/development/meshtastic/meshtastic-esp32/src/screen.cpp:788
0x400dde03: OneButton::tick(bool) at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tbeam/OneButton_ID1260/src/OneButton.cpp:150
0x400ddf0e: OneButton::tick() at /home/kevinh/development/meshtastic/meshtastic-esp32/.pio/libdeps/tbeam/OneButton_ID1260/src/OneButton.cpp:150
0x400ee4a1: loopTask(void*) at /home/kevinh/.platformio/packages/framework-arduinoespressif32@src-dfce4f8fb340e7e3eb7d44dd50813438/cores/esp32/main.cpp:19
0x40095376: vPortTaskWrapper at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ 
```

```
Guru Meditation Error: Core 1 panic’ed (LoadProhibited). Exception was unhandle d.
Core 1 register dump:
PC : 0x4017353b PS : 0x00060b30 A0 : 0x800dccbc A1 : 0x3f fd2a80
A2 : 0x00000001 A3 : 0x0000007f A4 : 0x0000001e A5 : 0x00 00007f
A6 : 0x00000000 A7 : 0x00000000 A8 : 0x00000180 A9 : 0x00 000040
A10 : 0x00000003 A11 : 0x00060b23 A12 : 0x00060b20 A13 : 0x3f fbac6c
A14 : 0x00000000 A15 : 0x00000000 SAR : 0x0000001a EXCCAUSE: 0x00 00001c
EXCVADDR: 0x00000180 LBEG : 0x400029ac LEND : 0x400029cb LCOUNT : 0x00 000000

ELF file SHA256: 0000000000000000

Backtrace: 0x4017353b:0x3ffd2a80 0x400dccb9:0x3ffd2aa0 0x400dbd17:0x3ffd2ac0 0x4 00d6806:0x3ffd2ae0 0x400dde03:0x3ffd2b00 0x400ddf0e:0x3ffd2b20 0x400d6a3b:0x3ffd 2b40 0x400ee4a1:0x3ffd2b70 0x40095376:0x3ffd2b90
```

geeksville, 2020-07-08:
hmm - works for me and the code looks correct to me.  Puzzling.  W

aHVzY2g, 2020-07-12:
it's also working for me (ttgo-lora32-v1)

geeksville, 2020-07-13:
alas - closing as can-not-reproduce.  Will wait to see if other reports arrive.

closed 2020-07-13 by geeksville