meshtastic/firmware #265: Huge latency on Bw500Cr45Sf128 config

ddv2005, 2020-07-08:
Hello,

I can't get message rate more than 1 packet per 1-2 seconds even on Bw500Cr45Sf128 config. I have 2 devices (TBEAM v1.1) on COM6 & COM7. 
Send script on COM6 device:
```
import meshtastic
interface = meshtastic.StreamInterface("COM6") 
for x in range(6):
  interface.sendText(str(x),'^all', False, False) 
```
COM7 device monitored by meshtastic.exe --device COM7 --debug. See attached m.log.
I am expecting that COM7 device will receive all 6 messages during 1 seconds. But actually it take 1-2 seconds for every message.

[m.log](https://github.com/meshtastic/Meshtastic-device/files/4888454/m.log)

Device settings:
```
Connected to radio
my_node_num: 2988741000
has_gps: true
num_channels: 13
region: "unset"
hw_model: "tbeam"
firmware_version: "unset"
packet_id_bits: 32
current_packet_id: 1247858935
node_num_bits: 32
message_timeout_msec: 300000
min_app_version: 172

preferences {
  position_broadcast_secs: 900
  send_owner_interval: 4
  wait_bluetooth_secs: 120
  screen_on_secs: 300
  phone_timeout_secs: 900
  phone_sds_timeout_sec: 7200
  mesh_sds_timeout_secs: 7200
  sds_secs: 31536000
  ls_secs: 3600
}
channel_settings {
  tx_power: 23
  modem_config: Bw500Cr45Sf128
  psk: "\327\241i\317\275\025y\301F\273t:\217\346@F\371\344\345\274\371D\336\007W\335\001i\336~Nc"
  name: "G1"
}
```

Professr, 2020-07-08:
LoRa is very slow, and a packet every 1-2 seconds doesn't seem unusual to me.  Is there a reason you expected it to be faster ?

ddv2005, 2020-07-08:
Raw LoRa bandwidth on Bw500Cr45Sf128 config about 21Kbps that about 28.224ms for 60 bytes payload.

geeksville, 2020-07-08:
Probably our random anti-collision transmit delays need to scale based on expected packet airtime requirements: https://meshtastic.discourse.group/t/data-bandwidth-and-latency/654/7

geeksville, 2020-11-14:
dup of #513 and fixed there

closed 2020-11-14 by geeksville