meshtastic/firmware #267: Pairing cubersome and gets lost frequently

palbeda, 2020-07-09:
I am using Meshtastic on an Android 10 device and a TTGO ESP32-v1 V0.7.10 and all earlier versions

I note that if I pair the TTGO on my phone, Meshtastic does not see paired device. Sometimes the name of the device is shown grewed out in the app, you cannot select it. Sometimes re-pair helps, usually not. 

Sometimes after clearing the apps memory it suddenly connects.

Pairing is sometimes lost when the TTGO hits the sleep mode.

When I have switched off the phone or walked away, thus loosing connectivity between Android and the TTGO, it does not seems to reconnect the already paired TTGO.

geeksville, 2020-07-09:
Oh!  Bummer - sorry about that.  What model of phone, Android version and
app version?

(Sent from a phone - please ignore typos)

palbeda, 2020-07-09:
Hi Kevin,Umidigi F2 android 10 TTGO ESP32-v1 V0.7.10No other Bluetooth issues that I am aware of Best regards,Peter--Sent from my Android phone with GMX Mail. Please excuse my brevity.

geeksville, 2020-07-09:
Hi Peter.  Thanks - One more question (sorry):

What version # of the android app?  You can find this in android settings / Apps / Meshtastic.

palbeda, 2020-07-09:
0.7.93--Sent from my Android phone with GMX Mail. Please excuse my brevity.

geeksville, 2020-07-12:
Thanks!  I'm going to merge this with #266 and track the fix there.

closed 2020-07-12 by geeksville