meshtastic/firmware #269: Show hash suffix & speed code at end of channel name

geeksville, 2020-07-12:
to prevent user channel confusion.  per forum post.

geeksville, 2020-07-18:
The rule for suffix will be:

channelname-XY

Where X is a letter from A-Z (base 26) representing a hash of the PSK for this channel - so that if the user changes anything about the channel (which does force a new PSK) this letter will also change.  Thus preventing user confusion if two friends try to type in a channel name of "BobsChan" and then can't talk because their PSKs will be different.

This also allows the option of someday if people have the PSK off (zero), the users COULD type in a channel name and be able to talk.

Y is a lower case letter from a-z that represents the channel 'speed' settings (for some future definition of speed)

cyclomies, 2020-07-20:
Related to the PSK: users could be allowed, to enter their own PSK at will (instead of a automatically generated PSK)?

commit 55dafcbe (2020-08-12): fix #269 see below

/**
 * Generate a short suffix used to disambiguate channels that might have the same "name" entered by the human but different PSKs.
 * The ideas is that the PSK changing should be visible to the user so that they see they probably messed up and that's why they
their nodes
 * aren't talking to each other.
 *
 * This string is of the form "#name-XY".
 *
 * Where X is a letter from A to Z (base26), and formed by xoring all the bytes of the PSK together.
 * Y is not yet used but should eventually indicate 'speed/range' of the link
 *
 * This function will also need to be impl [... cut]
closed 2020-08-12 by geeksville