meshtastic/firmware #272: Brightness slider activates on TTGO-LoRa32-V2.1 with serial Puttty connection.

thomslik, 2020-07-12:
Anytime TTGO-LoRa32-V2.1 is connected with a serial connection with Putty, the brightness slider activates and repeatedly  cycles from low to high. Toggling the screen by pulling the GPIO0 low to ground does not stop it. Ending the Putty session returns the unit to normal operation.

aHVzY2g, 2020-07-13:
I guess the issue can be closed https://meshtastic.discourse.group/t/alpha-tester-thread-please-try-device-code-0-7-11/298/179?u=ahvzy2g ?

thomslik, 2020-07-13:
> I guess the issue can be closed https://meshtastic.discourse.group/t/alpha-tester-thread-please-try-device-code-0-7-11/298/179?u=ahvzy2g ?

This is a totally unrelated issue. It has nothing to do with the T-Beams. It seems to be caused by different pin outs between the TTGO-LoRa32-V2.1 and V2.0.

aHVzY2g, 2020-07-13:
could you verify that this is the version you have https://github.com/LilyGO/TTGO-LoRa32-V2.1 , I will try to add support for it.

geeksville, 2020-07-13:
oh yes good point.  slider going up endlessly means the GPIO assignment for the button on the lora32-v2.1 is wrong.

aHVzY2g, 2020-07-13:
@thomslik please test #274 and let me know if it works

thomslik, 2020-07-13:
> could you verify that this is the version you have https://github.com/LilyGO/TTGO-LoRa32-V2.1 , I will try to add support for it.

Yes. That is the version I have.

thomslik, 2020-07-13:
> @thomslik please test #274 and let me know if it works

Sorry for my ignorance, but where do I find the bin to try?

thomslik, 2020-07-14:
Updated earlier reply, to include tried to build.

aHVzY2g, 2020-07-14:
@thomslik I will send you a bin asap

aHVzY2g, 2020-07-14:
Here you go I have two separate bin files. Please test both an tell me if they work. Before each flash use the [esp download tool](https://www.espressif.com/sites/default/files/tools/ESP32%26ESP8266_RF_Performance_Test_EN.zip) to erase the flash of your board.

[build.zip](https://github.com/meshtastic/Meshtastic-device/files/4918400/build.zip)

thomslik, 2020-07-14:
Thanks. Both versions still do the same. The unit appears to be working fine until the serial connection. Then the same problem with the brightness slider.

aHVzY2g, 2020-07-14:
[build.zip](https://github.com/meshtastic/Meshtastic-device/files/4918553/build.zip)
sry my mistake, did a copy-paste error

closed 2020-07-14 by thomslik

geeksville, 2020-07-14:
Thanks y'all!

(Sent from a phone - please ignore typos)

commit 44749470 (2020-07-16): fix #272 add support for tlora-v2-1-1.6
commit 4ee35a06 (2020-07-16): Merge pull request #280 from aHVzY2g/master

fix #272 add support for tlora-v2-1-1.6