meshtastic/firmware #284: loss of GPS lock not updated in the GUI

geeksville, 2020-07-15:
but the problem isn't the GUI.  It is that we only publish GPS updates when we have lock.  Once we lose lock we stop publishing.  Really - we should do one last publish when the state changes from locked to not locked (so the rest of the app establishes the new proper !locked state)

From this forum report:

geeksville, 2020-07-15:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/bug-tbeam-never-returns-to-no-sats-state/741/2

Professr, 2020-07-15:
Mind if I fix this one?

geeksville, 2020-07-15:
Feel free to fix any one you want! ;-)

Professr, 2020-07-15:
Hmm.  NEMA already looks ok, but UBlox always reports its last known fix.  Fortunately it seems to update the number of satellite locks it has, so I integrated that into the logic that determines whether the module has a lock or not.  It takes 5-10 minutes to really pick up on the loss of lock, but it does go back to "No sats" at that point.

PR #285 (2020-07-16): UBlox GPS now detects loss of lock
Fixes #284 

The UBlox GPS library caches its last known fix, but it does update the number of satellite locks returned by `getSIV()`
I've integrated `numSatellites` into the calculation for whether or not a lock exists, and changed the loop to always check the cached position and number of satellites regardless of the reported fix type.
closed 2020-07-16 by geeksville