meshtastic/firmware #293: Reboot loop with firmware-tlora-v2-1-1.6-US-0.8.1.bin

thomslik, 2020-07-18:
The new firmware-tlora-v2-1-1.6-US-0.8.1.bin is apparently different from the proper test bin which aHVzY2g created. The test bin works fine. This bin results in a continuous reboot. I reloaded the test bin and it's still good.

geeksville, 2020-07-18:
oops thanks problem was:

E (397) cpu_start: Failed to init external RAM!
abort() was called at PC 0x4008339d on core 0

kevinh@kevin-server:~/development/meshtastic/meshtastic-esp32$ bin/exception_decoder.py -e release/latest/elfs/firmware-heltec-US-0.8.1.elf ~/tmp/ex
stack:
0x4008f617: invoke_abort at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x4008f99d: abort at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x4008339d: call_start_cpu0 at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:205 (discriminator 1)

geeksville, 2020-07-18:
will affect any board without external ram.  I'll fix.

commit 20ac8d71 (2020-07-19): make PSRAM optional - new flag in ESPIDF - fix #293
commit ece75d1d (2020-07-24): make PSRAM optional - new flag in ESPIDF - fix #293
closed 2020-07-19 by geeksville