meshtastic/firmware #302: TBEAM 0.7 No GPS Satilites

DylanHoen, 2020-07-28:
I'm trying Meshtastic for the first time.
I put firmware-tbeam0.7-US-0.9.1.bin on 2 TBeams that have T22_V07 20180711 written on them.
I have 2 of them and can send messages between 2 cell phones with them.
They both say No Sats, even after being outside in the open on a sunny day for 15 minutes.
The android app (0.9.01), puts nothing on the map, and question marks for battery level.
At the moment, I see a question marks for signal levels in the android app, but the second cell phone has gone to work with its owner at the moment, so it is out of Bluetooth range.  I think I might have seen a signal level back when I had both cell phones, but I can't quite remember.
My cell phone is a Galaxy Note 8, US version.
I think the other cell phone might have been a Galaxy A70.
I tried testing with an old Galaxy S5, but I couldn't get the app to allow me to enter my name.

geeksville, 2020-07-28:
Hi!

? for battery level is a known bug for that board https://github.com/meshtastic/Meshtastic-device/issues/279

@slavino do you have any theories about why his board would say no satellites?

@DylanHoen Are you able to hook a serial app up to the board (921600 baud) and then press reset on the board and capture all the gunk it prints?  That info would really help track down what the problem is.

DylanHoen, 2020-07-28:
Does ESPHome-Flasher-1.2.0-windows-x64.exe count?
I clicked view logs after hitting reset, and it had:

Using 'COM8' as serial port.
Showing logs:
[16:07:53]ets Jun  8 2016 00:22:57
[16:07:53]
[16:07:53]rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
[16:07:53]configsip: 0, SPIWP:0xee
[16:07:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:07:53]mode:DIO, clock div:2
[16:07:53]load:0x3fff0018,len:4
[16:07:53]load:0x3fff001c,len:1044
[16:07:53]load:0x40078000,len:8896
[16:07:53]load:0x40080400,len:5828
[16:07:53]entry 0x400806ac
[16:07:54][D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

DylanHoen, 2020-07-28:
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒Emitting reboot packet for serial shell
▒▒Hbooted, wake cause 0 (boot count 1), reset_reason=reset
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=0.9.1, hwver=1.0-US
Setting random seed 2503406823
Total heap: 265728
Free heap: 238540
Total PSRAM: 4194252
Free PSRAM: 4194252
NVS: UsedEntries 88, FreeEntries 542, AllEntries 630
Turning on screen
Read RTC time as 0 (cur millis 107) valid=0
Connected to UBLOX GPS successfully
RadioConfig reset!
Installing AES128 key!
Initial packet id 7605993, numPacketId 4294967295
Loading saved preferences
Warn: devicestate is old, discarding
RadioConfig reset!
Installing AES128 key!
Installing AES128 key!
NODENUM=0x3ab94450, dbsize=1
Starting meshradio init...
Set radio: name=Default, config=3, ch=6, power=23
RF95 init result 0
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
GPS fix type 0
Ignoring invalid GPS month=6, year=19389, unixtime=-1
New GPS pos lat=0.000000, lon=0.000000, alt=0, pdop=320.400000, heading=0.000000, sats=0
Screen: Started...
sending owner !807d3ab94450/Unknown 4450/?50
Update DB node 0x3ab94450, rx_time=0
old user !807d3ab94450/Unknown 4450/?50
updating changed=0 user !807d3ab94450/Unknown 4450/?50
Adding packet record (id=0x00740eed Fr0x50 To0xff, WantAck0, HopLim3 Payload:User)
enqueuing for send (id=0x00740eed Fr0x50 To0xff, WantAck0, HopLim3 encrypted)
txGood=0,rxGood=0,rxBad=0
Starting low level send (id=0x00740eed Fr0x50 To0xff, WantAck0, HopLim3 encrypted)
showing standard frames
Transition powerFSM transition=boot timeout, from=BOOT to=ON
                                                            Setting bluetooth enable=1
Pre BT: 223436 heap size
                        Starting bluetooth
BLE task running
registered service 0x1800 with handle=1
registering characteristic 0x2a00 with def_handle=2 val_handle=3
registering characteristic 0x2a01 with def_handle=4 val_handle=5
registered service 0x1801 with handle=6
registering characteristic 0x2a05 with def_handle=7 val_handle=8
registered service 6ba1b218-15a8-461f-9fa8-5dcae273eafd with handle=10
registering characteristic f75c76d2-129e-4dad-a1dd-7866124401e7 with def_handle=11 val_handle=12
registering characteristic 8ba2bcc2-ee02-4a55-a531-c525c5e454d5 with def_handle=13 val_handle=14
registering characteristic ed9da18c-a800-4f66-a670-aa7547e34453 with def_handle=15 val_handle=16
FromNum handle 16
registered service cb0b9a0b-a84c-4c0d-bdbb-442e3144ee30 with handle=18
registering characteristic e74dd9c0-a301-4a6f-95a1-f0e1dbea8e1e with def_handle=19 val_handle=20
registering characteristic e272ebac-d463-4b98-bc84-5cc1a39ee517 with def_handle=21 val_handle=22
registering characteristic 4826129c-c22a-43a3-b066-ce8f0d5bacc6 with def_handle=23 val_handle=24
registering characteristic 5e134862-7411-4424-ac4a-210937432c77 with def_handle=25 val_handle=26
update result handle 26
Addr type 0, Private=0, Device Address: 69:80:7d:3a:b9:44

connection established; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=00:80:7d:3a:b9:44
 our_id_addr_type=0 our_id_addr=00:80:7d:3a:b9:44
 peer_ota_addr_type=0 peer_ota_addr=00:b4:bf:f6:7e:f2
 peer_id_addr_type=0 peer_id_addr=00:b4:bf:f6:7e:f2
 conn_itvl=30 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

[... 123 lines cut ...]
BLE fromRadio called omlen=1, ourlen=0
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932
GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431932

DylanHoen, 2020-07-29:
I was reading through the above log when I saw the line:
Meshtastic swver=0.9.1, hwver=1.0-US
I guess that means my board is version 1.0?
I installed:
firmware-tbeam-US-0.9.1.bin
on both boards.
I see some black lines through the text on one of the screens, but it still appeared to be working.
One got gps lock and a name appeared on the map on both cell phones (same name).
I ended swapping TBeams between the 2 cell phones, but still the same name appeared on the map on both phones.
Sometimes it is a bit confusing to figure out which TBeam was connected to which cell phone.
I tried swapping a usb charging cable between one TBeam and the other.
I think there is a long time delay on the charging indicator (in the android app).
At one point, I saw both of them show connected to external power at the same time (in the android app).
Eventually, after half an hour or so, both GPS locations showed up on both cell phone maps.
I appended a 2 to the end of the names, and the 2 propagated to one of the android ap maps for both names, but after 15 minutes, the 2 had only propagated to one of the names on the other android app map.
The people list names match the map names.
Both TBeams say no GPS on one of their screens, but they have a compass arrow on another screen.

Do I really have a TBeam 1.0, or was I supposed to wait an hour to get a GPS lock in the 0.7 firmware?

Did an Aliexpress vendor make a 1.0 board, but keep the old V07 text on the board?

Does it use the cell phone GPS if the TBeam GPS is broken?

DylanHoen, 2020-07-29:
I had a couple questions at the end of my previous post that I didn't want to hide by posting this new post, but I had another observation:
I took the 2's off of the end of the names.  I managed to get the cell phone app into a state where the names were updated in the list of people, but, on the map, the current cell phone's name was updated, but the name from the remote cell phone wasn't updated, and still had a 2.
I checked a few minutes later, and this was fixed.  I'm guessing that GPS messages come through every once in a while, but the name change message came through right away, and the map was finally updated when a new GPS message came through.
I looked at the external power indicator on the list of people in the android app, and both TBeams are listed as externally powered on both cell phones.  One of them hasn't had the usb cable connected in at least 20 minutes.  I sent a text message from each cell phone to the other, but the external power indicator still hasn't been updated.

slavino, 2020-07-29:
I'm currently updating to latest builds (didn't have time recently to touchbase with all the changes).

In general the T-Beam v0.7 labeled (T22_V07 \n 20180711) should get this firmware firmware-tbeam0.7-EU433-0.9.1.bin (or adequate region/freq). I'm not able to move outside immediately and do clear sky test so I'm getting the invalid GPS message as well.

And yes - GPS pins differ between T-Beam 0.7 and later versions and there are also other diffs.

geeksville, 2020-07-29:
re: external power yes - there is a bug in tbeam 0.7 power indications.  #279 - I'm fixing it soon.

closed 2020-07-29 by geeksville