meshtastic/firmware #303: TTGO-LoRa32-V2.1 battery level display continuously shows charging

thomslik, 2020-07-29:
TTGO-LoRa32-V2.1, the battery level screen display shows only the charging symbol, not the battery level, while running with a battery attached. Installed my new batteries and discovered this, after running on just battery power.

geeksville, 2020-07-29:
alas - this device doesn't have a charging and battery voltage sensor (I don't think - @slavino do you happen to know if battery voltage is hooked to any of the CPU analog inputs?  I don't have one of these boards)

slavino, 2020-07-30:
There are more v2 boards

- https://github.com/LilyGO/TTGO-LORA32/blob/master/BaT_test_ssd1306/BaT_test_ssd1306.ino
- https://github.com/LilyGO/TTGO-LoRa32-V2.1

As they marked exactly the same way the pin 35 , it may be worth giving it a try.
http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1271&FId=t3:50003:3

@geeksville **And Pax counter shows that it should be pin 35...**
https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/hal/ttgov21new.h

geeksville, 2020-07-30:
@slavino ooh - that is super useful.  I'll add checking that ADC when I add tbeam 0.7 support.

geeksville, 2020-08-10:
Merging with #279

closed 2020-08-10 by geeksville