meshtastic/firmware #304: Ignoring invalid gps

MeisterQ, 2020-07-31:
**Describe the bug**
I have 2 similar tbeam boards, i flashed firmware-tbeam-EU865-0.9.1.bin and paired with my phone. Now i can see 2 devices on both phones, but on the 2nd one it always says on Log:

GPS fix type 0
Ignoring invalid GPS month=0, year=1336, unixtime=1297431933

Always and always... the blue LED on the board is flashing like at the first (Where GPS is working).

But on the first device the LED below the gps neo module is flashing red.

But on the second the GPS Neo led is not flickering.

I tried restarting and reflashing the device a couple of times, and the error came again. 

On app at both phones i can see both devices.

Well. What to do?

**Device info:**
 - Device model: TBeam
 - Software Version 0.91

**Smartphone information (if relevant):**
 - Device: Samsung Galaxy S7
 - OS: Android 8.0.0
 - App Version 0.9.01

slavino, 2020-08-01:
T-Beam has more versions (not only GPS pins are different to later versions).
The one labeled "T22_V07 20180711" requires another firmware because it is referenced as 
**firmware-tbeam0.7-EU433-0.9.1.bin**
**firmware-tbeam0.7-EU865-0.9.1.bin**
etc.

geeksville, 2020-08-10:
@MeisterQ is it possible your boards are 0.7 tbeams?  see comment from @slavino above.