meshtastic/firmware #305: Firmware 0.9.1 update issue with Heltec LoRa 32 v2 devices

z0ne0, 2020-07-31:
Please - if you just have a question (i.e. not a bug report or a feature proposal), post in our [forum](https://meshtastic.discourse.group/) instead.

**Describe the bug**
Both 1 & 2 below occurs while flashing `firmware-heltec-EU865-0.9.1.bin` release.

1- Update via esphome-flasher-1.2.0 on macbook (10.15.6 Catalina)
Flashing heltec LoRa 32 v2 device for firmware v0.9.1 outputs following log at the end:

```
Showing logs:
[20:13:32]E (434) spiram: SPI RAM enabled but initialization failed. Bailing out.
[20:13:34] (\~
```

Even though each time it outputs this log, firmware update is sometimes successful. Only way to understand this is to check display (top right corner) on restart.

2- OTA via BLE on iOS 
At the end of transferring all chunks of firmware file, serial monitor outputs the log below and device restarts. 
```
Guru Meditation Error: Core 1 panic’ed (Cache disabled but cached memory region accessed.
```

**To Reproduce**
1st one: just try to flash `firmware-heltec-EU865-0.9.1.bin`file to a heltec LoRa 32 v2 device and check "Showing logs" section of logs at the end of flashing process.

2nd one: unfortunately, iOS client is not ready to publish yet

**Device info:**
 - Device model: Heltec LoRa 32 v2
 - Software Version: 0.9.1

**Smartphone information (if relevant):**
 - Device: iPhone XS
 - OS: 13.4.1 & 13.6
 - App Version: Dummy iOS client 

**Additional context**
The same esphome-flasher and iOS client works fine with firmware 0.7.9 and 0.7.10.

geeksville, 2020-08-02:
@zone re subissue 2
Do you remember what version of the firmware you were upgrading **from**?  Thanks much

I'm working on subissue 1 right now

geeksville, 2020-08-02:
re subissue 1
Does your board operate properly (ignoring firmware update issues) with 0.9.1?  i.e. connecting to the phone, updating display etc?

Because I just checked and that message is (by itself) harmless.

z0ne0, 2020-08-03:
subissue 2:
Tried with 3 heltec boards (exact same versions) each having different meshtastic firmware.
Upgrading to 0.9.1 from 0.7.9, 0.7.10 and stock heltec firmware (the very first use of this board).

subissue 1:
ignoring fw update issue with 0.9.1: connecting to the phone, updating display is ok. 
Even though I think it's not related to update issue, what I noticed is discovering the devices takes a bit longer. And sending message from phone through device A and receiving from device B (LoRa message) takes longer than 0.7.10 e.g. 4-8 seconds, so it's not immediate although devices almost next to each other (30 cm, signal strength shows %100)

Working on pairing, sending and receiving message and OTA, I'm not sure if there are other things operating differently or unexpectedly.

Alas it outputs correct state for firmware upgrade (OTA done, failed, interrupted etc), that's sth to avoid misleading as well.

z0ne0, 2020-08-03:
subissue 2:

good news:
I tried to do everything from scratch (flashed the devices with fw 0.7.9, rebooted macbook etc) and couldn't reproduce the error. That's good I guess, since tried many patterns to reproduce. I am able to upgrade fw to 0.9.1 via BLE on iOS.

kinda bad news: 
When upgrade to 0.9.1 finishes, device doesn't return upgrade result characteristic value and restarts. So, iOS client doesn't know what to do or handle. For example: device needs to be re-paired. If upgrade is successful, inform user to re-pair; if not, reconnect automatically. 

If you don't have other findings, by making sure upgrade result always returned (if not a forced restart), subissue 2 is done I think. Thank you! (Btw about re-pairing need after fw upgrade: is it a limitation or sth in the to-do?)

geeksville, 2020-08-04:
re: upgrade to 0.9.1 forces repair
yes, alas - this is a one time limitation of switching from the old BLE stack to the new one.  So I think we'll have to live with it;-)

re: couldn't reproduce
I was able to reproduce, it was a bug that could occur if the node receives an interrupt (because of a RX packet or button press) while doing a software update.  I'm going to retitle this bug for the fix.

closed 2020-12-31 by geeksville