meshtastic/firmware #308: Meshtastic protocol

bobricius, 2020-08-04:
Is somewhere described Meshtastic protocol/ packet structure?
There is lot of source files, where I can found packet definition?

cyclomies, 2020-08-05:
@geeksville will update the documentation regarding to the protocol, whenever he has the time for it. If I had to guess, this might happen within this month, as it is not the highest on the working queue. 

I will open a discussion of the protocol, and other underlying factors, when the documents are updated.

geeksville, 2020-08-10:
@bobricius the packets themselves are protobufs described in the meshtastic-proto repo.  I took two weeks off but back at things today, I'll check in a big update to mesh-alg.md this afternoon which @cyclomies kindly offered to improve.

qurm, 2020-09-28:
Hi @geeksville, I started to document the protobufs, based on the mesh.proto file.  This was partly to help me get familiar with the protocols, but I'd like to contribute a document that helps to resolve this issue.  Maybe review this file, and give me some feedback - if it is useful I'll raise a pull request.  Not sure which repo you'd prefer these docs to be in?
[https://github.com/qurm/Meshtastic-protobufs/blob/master/protobufs.md](https://github.com/qurm/Meshtastic-protobufs/blob/master/protobufs.md)

closed 2021-01-15 by geeksville