meshtastic/firmware #309: Message confirmations not working correctly

mikeberry, 2020-08-04:
Let me start by saying I really love what you did with meshtastic! I followed it since 0.6 and the improvements are enormous and the speed at which you tackle issues is great! I made a test today with 5 nodes 1x heltec, 4x ttgo t-beam 1.1 ([https://de.aliexpress.com/item/33047631119.html?spm=a2g0s.9042311.0.0.79f74c4dhCjubb](https://de.aliexpress.com/item/33047631119.html?spm=a2g0s.9042311.0.0.79f74c4dhCjubb)) . We walked around the city (old town, 5 story buildings about 1.4 km radius, some hills) and sent messages to each other for about 1 hour.
Our main issue was that we could never be sure if the message was received or not. Sometimes there would be a crossed out cloud even if all the nodes received the message. Sometimes there would be a checked cloud even if only a few nodes got the message. It is not a problem if a message cannot be sent at some point but the sender has to know reliably.

Also it would help to know which users received the message in addition to just the ticked cloud. Like in whatsapp where you can longpress on the message and then information.

geeksville, 2020-08-10:
I think this is an odd side effect of the same thing that causes: https://github.com/meshtastic/Meshtastic-Android/issues/86

I'll be working on that bug soon.  THanks for the report.

mikeberry, 2020-08-10:
Do you think it is an android or device issue?

geeksville, 2020-08-15:
@mikeberry android.  Though I've fixed #86 now so we should see if htis is still a problem.

sachaw, 2021-03-31:
@geeksville Is this now resolved?

geeksville, 2021-04-01:
yep - thanks!

closed 2021-04-01 by geeksville