meshtastic/firmware #312: Meshtastic User Groups

chron0, 2020-08-10:
Sorry if this isnt the best place to suggest this, but here it goes anyways:

Playing, debugging, testing and extending mesh networks is much more fun and feasible when not doing it alone. I love the idea of this kind of mesh network, similar to the basic idea of freifunk etc. so I think it would be great if we would try to get some local user groups going in our areas.

I'm giving it a start with: Munich, Germany

geeksville, 2020-08-10:
@chron0 I think that's a great idea.  How about I make a "Local users group" section on the forum?  And then you can make a thread for Germany?

geeksville, 2020-09-17:
Created a regional topic area (linked above).  Sorry it took so long.

closed 2020-09-17 by geeksville