meshtastic/firmware #317: 923mhz band for AU/NZ

shanemdowd, 2020-08-13:
Hi there, very interested to have the 923Mhz bands included in upcoming releases for the Australia/New Zealand regions.

See https://meshtastic.discourse.group/t/hello-and-success/861/5

Thanks
Shane

geeksville, 2020-09-08:
@shanemdowd  Finally working on this:

Since meshtastic uses Lora (and doesn't need to talk to standard lorawan gateways) I think based on this comment:

https://www.thethingsnetwork.org/forum/t/frequency-plan-new-zealand/6846/2

That using the wider US 915 plan is legal in NZ and will give you better performance (because more channels).  Any objection to me just changing the readme to link this comment and recommend for NZ to use the existing US settings?

shanemdowd, 2020-09-09:
@geeksville This works fine for me thanks, did you have any thoughts on using meshtastic for monitoring nodes, ie status of GPIO for example?

geeksville, 2020-09-10:
@shanemdowd oh yes - work should start on that soonished (possibly within a month, definitely within two).  good thread here  if you want to chime in:

https://meshtastic.discourse.group/t/extension-of-the-functionality-of-devices-meshtastic/411/2

closed 2020-09-10 by geeksville