meshtastic/firmware #320: Unable to get T-Beam/s to join mesh

Keristero, 2020-08-15:
I received my T-Beams today, ordered from here https://www.aliexpress.com/item/33047631119.html?spm=a2g0s.9042311.0.0.67534c4d0mpfJV

I've flashed them with the `firmware-tbeam-EU865-0.9.1` firmware
I've paired both devices to android phones, I've set them to use the same channel using the qr code.

On each app I can see the username I specified for that phone, but I never see both nodes. They both eventually get GPS.

in the console one of them was spamming
`can not send yet, busyTx`
after a while it started spamming
`radio wait to sleep, txEmpty=0` at the same time

The other one is just logging battery level, GPS fix type 0, and Bluetooth traffic.
I have reflashed them a few times, and tried with another android phone...
I'm just trying to get them to join the mesh so I can try out the messaging and do a range test.

Tried with these phones, they pair fine - I don't think it is a phone issue
Xiaomi Redmi note 4x
Xiaomi Mi 9t
Xiaomi Redmi note 6

geeksville, 2020-08-15:
hmm - interesting.  could you post a full log from the node that was pringing "can not send yet" (that is normal to occur for a few times occasionally (while waiting for a quiet network) but not normal to say a lot.

Keristero, 2020-08-16:
Right, I understand it is optional with lora to check if there is traffic before sending a tx, I'll send a log when I get a chance to try again. Perhaps I can try disabling the check too?

geeksville, 2020-08-16:
no way to disable the check (and you wouldn't want that anyways - because that would guarantee your transmitted packet would corrupt someone elses - and they would both be lost)

geeksville, 2020-08-29:
(alas - closing to keep the bug queue tidy, if this is still a problem please add a log and reopen)

closed 2020-08-29 by geeksville

Keristero, 2020-09-02:
Quick follow up, the board which was giving me errors has since completely died, I think this may have just been a hardware issue.