meshtastic/firmware #321: Publish APK on F-Droid

Alixpat, 2020-08-15:
Please - if you just have a question (i.e. not a bug report or a feature proposal), post in our [forum](https://meshtastic.discourse.group/) instead.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device info:**
 - Device model: [e.g. TBEAM]
 - Software Version [e.g. 0.7.8]

**Smartphone information (if relevant):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App Version [e.g. 0.7.2]

**Additional context**
Add any other context about the problem here.

closed 2020-08-15 by Alixpat

insanity54, 2021-04-10:
I would like to see Meshtastic on F-Droid

geeksville, 2021-04-11:
> I would like to see Meshtastic on F-Droid

Alas, that probably won't be happening soon.  I'm on mobile now but the meshtastic-android GitHub issues has a good description on why...

I will keep doing apk releases though.

insanity54, 2021-04-11:
Ah yes, thank you. I see the issue https://github.com/meshtastic/Meshtastic-Android/issues/6