meshtastic/firmware #322: Boot loop on low battery with TTGO LORA32 v2.1(1.6)

tschundler, 2020-08-16:
Originally reported by @thomslik in https://github.com/meshtastic/Meshtastic-device/issues/279#issuecomment-673534078 related to misreading the battery level. Even with correct battery measurement, when the battery gets low, the device goes into a boot loop.

To simulate with USB connected, I reverted [3de1607](https://github.com/geeksville/Meshtastic-esp32/commit/3de1607cea40320408eda6bea42729c01f6ce5f2), and here is the output:

```
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:732
load:0x40078000,len:9132
ho 0 tail 12 room 4
load:0x40080400,len:5100
entry 0x40080640
E (448) spiram: SPI RAM enabled but initialization failed. Bailing out.
Emitting reboot packet for serial shell
⸮⸮ Hbooted, wake cause 3 (boot count 6), reset_reason=reset
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=unset, hwver=unset
Setting random seed 1133865350
Total heap: 265572
Free heap: 237968
Total PSRAM: 0
Free PSRAM: 0
NVS: UsedEntries 183, FreeEntries 447, AllEntries 630
Using analog input for battery level
Turning on screen
Read RTC time as 7 (cur millis 120) valid=0
ERROR: No UBLOX GPS found
Hoping that NEMA might work
RadioConfig reset!
Installing AES128 key!
Initial packet id 2051513129, numPacketId 4294967295
Loading saved preferences
Loaded saved preferences version 11
Installing AES128 key!
NODENUM=0x1d698bf8, dbsize=1
Starting meshradio init...
Set radio: name=Default, config=3, ch=6, power=17
RF95 init result 0
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Trigger powerFSM 10
Transition powerFSM transition=LowBat, from=BOOT to=SDS
Entering deep sleep for 31536000 seconds
Turning off screen
Writing preferences

```

tschundler, 2020-08-16:
reason 3 is apparently `ESP_SLEEP_WAKEUP_EXT1` - https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#external-wakeup-ext1

tschundler, 2020-08-16:
disabling BUTTON_PIN in configuration.h avoids the boot loop, but then of course no working button.
I tried -DBUTTON_NEED_PULLUP, but it still bootloops.

LaurentLouf, 2020-08-16:
Could this be a brownout ? Do you have any mean to measure the power voltage ?

tschundler, 2020-08-16:
The detected voltage is deliberately wrong on the example above to force it to sleep. It's powered by USB with no battery for testing.

It only seems to occur when trying to sleep due to low battery. And the problem is clearly related to trying to use the RTC interrupts to wake up on button press, since without code related to button pins, it does go to sleep and stays asleep.

While I have had past issues with brownouts in other projects, with the ESP 32 it has reliably set the reset reason to RTCWDT_BROWN_OUT_RESET

geeksville, 2020-08-18:
Oh I know the problem! We are probably not setting the wake pin correctly
for lora32 - because the internal pull-up resistor is not turned on for
that board?!

(Sent from a phone - please ignore typos)

thomslik, 2020-08-23:
I added the requested line in configuration.h as shown in my file here:
#define LED_PIN 25     // If defined we will blink this LED
#define BUTTON_PIN                                                                                                               \
    12 // If defined, this will be used for user button presses, if your board doesn't have a physical switch, you can wire one
       // between this pin and ground
#define BUTTON_NEED_PULLUP
#define USE_RF95

I then modified power.cpp to simulate the low voltage condition. The unit still goes into a boot loop as with the following log file:
��␀␂H␁booted, wake cause 0 (boot count 1), reset_reason=reset
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=unset, hwver=unset
Setting random seed 1006200448
Total heap: 265664
Free heap: 238428
Total PSRAM: 0
Free PSRAM: 0
NVS: UsedEntries 91, FreeEntries 539, AllEntries 630
Using analog input for battery level
Turning on screen
Read RTC time as 0 (cur millis 92) valid=0
ERROR: No UBLOX GPS found
Hoping that NEMA might work
RadioConfig reset!
Installing AES128 key!
Initial packet id 1389939266, numPacketId 4294967295
Loading saved preferences
Loaded saved preferences version 11
Installing AES128 key!
NODENUM=0x91828afc, dbsize=1
Starting meshradio init...
Set radio: name=Default, config=3, ch=6, power=17
RF95 init result 0
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Battery 3mV 0%
Trigger powerFSM 10
Transition powerFSM transition=LowBat, from=BOOT to=SDS
Entering deep sleep for 31536000 seconds
Turning off screen
Writing preferences
��␀�␀���␀�␀␀␀�␀�����␀��␀␀␀␀␀␀␀␀␀␀␀␀␀�␀␀␀�␀␀␀␀␀␀␀␀�␀␀␀␀␀␀␀␀�␀���␀��␀��␀␀�␀��␀␀�␀␀␀��␀�␀��␀�␀␀␀␀����␀␀␀␀�␀␀␀���␀���␀�␀���␀��␀␀���␀���␀�␀�����␀␀����␀��␀����␀�␀����␀�␀␀�␀␀␀�������␀��␀�␀␀␀␀��␀␀�␀␀␀␀␀�␀�␀��␀���␀␀����␀␀��␀��␀��␀������␀␀�␀␀��␀��␀��␀␀����␀␀��␀��␀�␀����␀�␀��␀␀�␀�␀����␀��␀␀��␀���␀��␀��␀��␀�␀�␀␀␀�␀␀␀␀␀␀��␀��␀��␀���␀␀�␀␀␀␀␀�␀␀␀�␀��␀���␀␀�����␀��␀�␀␀␀␀���␀���␀���␀␀�␀�␀�␀���������␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀�␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀␀␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀�␀��␀�␀␀���␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀␀��␀␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀���␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀��␀␀�␀�������␀␀���␀�␀�␀␀␀��␀␀��␀�␀�␀␀␀�␀�␀������␀�����␀␀␀␀��␀��␀��␀�␀␀␀␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀��␀␀␀␀�␀␀�␀␀�␀␀␀␀�␀␀␀␀␀�␀␀����␀�␀��␀�␀␀�␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀��␀␀␀␀�␀␀�␀␀�␀␀␀␀�␀␀�␀�␀�␀␀����␀�␀��␀�␀��␀�␀␀␀␀␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀�␀�␀␀␀␀␀��␀␀␀␀␀␀␀␀␀␀�␀␀����␀�␀��␀�␀�␀␀�␀␀�␀␀␀␀␀��␀␀�␀␀�����␀␀␀␀␀␀␀���␀��␀��␀��␀␀�␀␀␀␀␀␀␀␀␀�����������␀␀␀�␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀�␀�␀␀␀␀␀␀␀␀␀␀�␀␀␀␀␀␀�␀␀����␀�␀��␀�␀���␀�␀␀␀␀␀␀��␀␀�␀��␀�␀��␀��␀␀��␀�␀␀␀␀␀�␀�␀␀␀␀␀␀␀␀␀␀�␀␀�␀␀␀��␀␀�␀��␀�␀␀␀�␀␀�␀␀�␀␀␀�␀�␀␀␀�␀�␀��␀��␀␀��␀�����␀�␀␀␀�␀��␀���␀␀�␀␀␀␀���␀���␀�␀␀��␀�␀���␀�␀␀�␀����␀�␀␀�␀␀␀␀�����␀��␀␀�␀��␀���␀��␀���␀���␀�␀���␀��␀���␀�␀���␀�����␀��␀␀␀␀��␀��␀��␀����␀�␀␀�␀�␀␀␀␀␀��␀��␀��␀���␀��␀���␀�␀␀�������␀��␀�␀��␀␀�␀�Emitting reboot packet for serial shell
��␀␂H␁booted, wake cause 3 (boot count 2), reset_reason=reset
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=unset, hwver=unset
Setting random seed 3664589159
Total heap: 265664
Free heap: 238428
Total PSRAM: 0
Free PSRAM: 0
NVS: UsedEntries 91, FreeEntries 539, AllEntries 630
Using analog input for battery level
Turning on screen
Read RTC time as 1 (cur millis 92) valid=0
ERROR: No UBLOX GPS found
Hoping that NEMA might work
RadioConfig reset!
Installing AES128 key!
Initial packet id 1883797361, numPacketId 4294967295
Loading saved preferences
Loaded saved preferences version 11
Installing AES128 key!
NODENUM=0x91828afc, dbsize=1
Starting meshradio init...
Set radio: name=Default, config=3, ch=6, power=17
RF95 init result 0
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
[... 15 lines cut ...]
Total PSRAM: 0
Free PSRAM: 0
NVS: UsedEntries 91, FreeEntries 539, AllEntries 630
Using analog input for battery level
Turning on screen
Read RTC time as 2 (cur millis 92) valid=0
ERROR: No UBLOX GPS found
Hoping that NEMA might work
RadioConfig reset!
Installing AES128 key!
Initial packet id 1223135943, numPacketId 4294967295
Loading saved preferences
Loaded saved preferences version 11
Installing AES128 key!
NODENUM=0x91828afc, dbsize=1
Starting meshradio init...
Set radio: name=Default, config=3, ch=6, power=17
RF95 init result 0
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Trigger powerFSM 10
Transition powerFSM transition=LowBat, from=BOOT to=SDS
Entering deep sleep for 31536000 seconds
Turning off screen
Writing preferences
��␀�␀���␀�␀␀␀�␀�����␀��␀␀␀␀␀␀␀␀␀␀␀␀␀�␀␀␀�␀␀␀␀␀␀␀␀�␀␀␀␀␀␀␀␀�␀���␀��␀��␀␀�␀��␀␀�␀␀␀��␀�␀��␀�␀␀␀␀����␀␀␀␀�␀␀␀���␀���␀�␀���␀��␀␀���␀���␀�␀

geeksville, 2020-08-25:
ok thanks - I need to read the datasheet today, I dimly remember some extra call is needed to leave that pullup powered during sleep (all of the other boards have an external pullup, which is why they don't show this problem)

geeksville, 2020-08-26:
oh - I think I found the problem (probably)!  we were isolating gpio12 (which isn't used on other boards) to save power during sleep.  gpio12 is the button for this board. @thomslik would you mind pulling this commit and seeing if it works better?

thomslik, 2020-08-26:
I think that's it! Loaded your version and simulated low battery with modified power.cpp. It goes to sleep as soon as it boots up. Pressing the added button (pin 12 to ground) immediately wakes it for a second before it returns to sleep. No more boot loop! 
Good job. Thanks!

geeksville, 2020-08-27:
yay!  thanks for your help!

closed 2020-08-27 by geeksville