meshtastic/firmware #343: Device restarts when phone joins channel, Android phone gets stuck in a loop trying to pair with it.

mc-hamster, 2020-08-27:
Device reboots when phone joins channel, Android phone gets stuck in a loop trying to pair with it.

Device A = T22_v1.1 20191212, Meshtastic 0.9.3
Device B = T22_v1.1 20191212, Meshtastic 0.9.3
Phone A = Blu, View 1, Android 9, Meshtastic 0.9.05
Phone B = Blu, View 1, Android 9, Meshtastic 0.9.05

Preconditions:

- Phone A is paired with Device A on the default channel.
- Phone B is paired with Device B on the default channel.

Steps to reproduce:

1. Phone A changes the channel settings on Device A to "oontz". 
2. Phone A sends the channel join link via email (Also reproducible using the QR code) to Phone B 
3. Phone B opens the email and when prompted, joins the channel sent by Phone A
4. Device B reboots itself and reverts back to the default channel (or the changes were never saved, I can't tell)
5. Phone B prompts to pair with the Bluetooth device of Device B. The dialog to pair comes back up every few minutes until I kill the Meshtastic application.
6. Phone B says "You haven't yet paired a Meshtastic compatible ..."

mc-hamster, 2020-08-28:
Update

Today, I was able to pair Phone B with Device B after leaving them turned on over night.

I'm now unable to reproduce the original problem.

mc-hamster, 2020-08-28:
I spoke too soon. The phone is now trying to pair with the device every few minutes again. The device shows a new pairing code the phone asks to pair.

mc-hamster, 2020-08-31:
More information ...

If a Meshtastic device is recently reset via firmware update and has only ever paired with one device, this problem doesn't seem to happen.

If after reset, the device pairs with Device A and then later pairs with Device B, then device B will exhibit the problem described in this report.

I seem to be able to reproduce this problem without having to change channels.

Tried this today with device 0.9.05 and android 0.9.06..

geeksville, 2020-09-02:
ok - I'll try to repro here with using the same phone to go back and forth between a single device.

mc-hamster, 2020-09-14:
Have not seen this in a while.

I'll open a new issue if this happens again.

closed 2020-09-14 by mc-hamster