meshtastic/firmware #352: Add extra fields (GPS, channel info etc...) to current settings page or add a second settings page

geeksville, 2020-08-31:
(from the linked thread)

You are absolutely right, it would be very useful.
It would be ideal to add (Power on device Time in minutes and hours under the GPS coordinates and the battery percentage)
I think it’s not so complicated for @geeksville to implement.
I also think adding a screen in the TTGO menu with technical parameters of the radio would be great.

1-GPS coordinates
2- Power on Time
3- Battery status in “mV” and “%”
4- Transmission mode selected
TX - ( “Fast Low~ Range” )
TX - ( “Fast Med.~ Range”)
TX - (“Slow Long~ Range”)
TX - (“Slow V-Long~ Range”)
5- Bluetooth “on” - “off” status

With some abbreviated strings everything can stay inside a single menu of the TTGO

geeksville, 2020-08-31:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/show-gps-coordinates-on-ttgo-mini-display/1146/5

mc-hamster, 2020-09-02:
I’ll throw my hat into the arena for this. 

Please give me a few days. My C is very rusty.

mc-hamster, 2020-09-02:
Was able to add a screen with some test text. That's enough for tonight.

geeksville, 2020-09-02:
super cool!  thanks for the help!

mc-hamster, 2020-09-03:
@geeksville You use the word "Trampoline" in Screen.h in the context of drawDebugInfoTrampoline.

What's the intent behind that word? Do you intend that to eventually become a dispatcher to other screens that display something other than information about nodes?

mc-hamster, 2020-09-06:
PR https://github.com/meshtastic/Meshtastic-device/pull/366

Tested on tlora-v1 and tbeam

Created a new screen with:

- Battery voltage, power source, charging status
- Time since reboot (known roll over bug @ 49 days)
- Display the radio "mode"
- Display status of the GPS module, or lack of module
- Display coordinates if we get a GPS lock

geeksville, 2020-09-07:
super cool @mc-hamster!  I'll merge it later today!

geeksville, 2020-09-07:
> @geeksville You use the word "Trampoline" in Screen.h in the context of drawDebugInfoTrampoline.
> 
> What's the intent behind that word? Do you intend that to eventually become a dispatcher to other screens that display something other than information about nodes?

Alas - I don't know. ;-).  I think that concept/note was added by either @girtsf or @Professr.

girtsf, 2020-09-10:
> Alas - I don't know. ;-). I think that concept/note was added by either @girtsf or @Professr.

Uh, I don't remember either what I was thinking, if that was me. :) From a quick glance, maybe because it gets the `Screen *` from the user data `void *` and then calls that instead?

mc-hamster, 2020-09-12:
Update with PR #380

mc-hamster, 2020-09-14:
Closed. It's in 1.0

closed 2020-09-14 by mc-hamster