meshtastic/firmware #359: Interop RFE - GPS geoidal height

a-f-G-U-C, 2020-09-03:
## Background
In geodesy, the height of a point can be represented in two different ways:
- orthometric or MSL - relative to the (real) mean sea level
- ellipsoidial or HAE - relative to an (idealized) reference ellipsoid

The difference between these two values is a datum called “geoidal separation” or “geoidal height”, basically a spot measure of the irregularity in the real shape of the Earth. Mathematically, HAE = MSL + geoidal_height.

The concept is explained briefly here: https://gis.stackexchange.com/questions/174046/ and beautifully here: https://www.esri.com/news/arcuser/0703/geoid1of3.html

Various geospatial applications prefer one reference system or the other, and modern GPS receivers  commonly include both the MSL altitude and the geoidal height value in the fix data to accommodate both preferences. 

## Currently
Meshtastic transmits the MSL altitude, but not the associated geoidal height. This makes the HAE altitude hard to calculate, and causes an incompatibility with HAE-oriented systems such as the CoT “Cursor-on-Target” protocol - which is the backbone of several applications relevant from an interop angle.

## Request
It would be nice if the Meshtastic protocol could include the geoidal height alongside the MSL, to support both MSL and HAE centric applications. The value is readily available from GPS, and generally requires the same treatment as the MSL altitude value. 

Thank you.

mc-hamster, 2021-03-27:
@a-f-G-U-C I'm going through old requests. How would you propose we use or display this information?

slavino, 2021-03-27:
How big is the observed difference between ublox implementation and NMEA? 
Does it still exist as issue 359 was resolved?

sachaw, 2021-03-27:
I think @a-f-G-U-C wants an additional property on Position messages, i.e geoidalSeparation

mc-hamster, 2021-03-27:
@slavino Dunno. We use the UBlox. We're now using MSL in the position packet.

We're now reading altitude with getAltitudeMSL() -- that was the fix in 363. HAE is read with getAltitude(). Just need to know how it's planned on being used.

Unless there's a specific use for it, we should stick to one or the other. It would increase airtime use for the position packet. It's also possible to have a plugin specifically or extended position messages that would include the various altitude calculations and have for cases where it's needed so it won't impact everyone else.

sachaw, 2021-03-27:
Agreed

a-f-G-U-C, 2021-03-27:
All, if the decision on HAE xor MSL is still up for discussion, **please**, consider using HAE preferentially, or even **HAE only**. It's the native measurement of the GPS protocol, while MSL is just an ugly hack for backward compatibility with pre-GPS datums, and IMO should be phased out from any real-life usage.
Also, as said before, modern applications like the CoT based apps are HAE only, and HAE support is a prerequisite of interfacing cleanly with these systems. I am not aware of any applications that prefer MSL.

My suggestion is:
- for max compatibility and future proofing, make HAE the standard elevation reference in Meshtastic (across all GPS models etc)
- for universal and backward compatibility, the geoidal separation (see top of page) *may be optionally included*, but most likely nobody will care

I have some relevant quotes on the subject which I may link in a separate comment when I find them.

a-f-G-U-C, 2021-03-27:
Found the quote that I was looking for, which explains the issue from an interop perspective:

> MSL is empirically derived (as is AGL from DTED). MSL is based on an estimate of “sea level” which is affected by the density of the Earth’s crust. “Sea level” is actually an irregular,
undulating surface that’s higher where the crust is densest. The shape of this surface is modeled by a big 2-D table of measurement, an Earth Gravity Model (EGM). The model is refined periodically [...]
In short, MSL is an interoperability nightmare. You have to have the same model, gridding, and interpolation to exchange information, or errors will result. All this is avoided if we use HAE. HAE is a mathematical model with just a few parameters for ellipse axes. It’s simple, accurate, and stable.

From [DTIC ADA637348: The Developer's Guide to Cursor on Target](https://archive.org/details/DTIC_ADA637348)

sachaw, 2021-03-30:
+1 for switch to HAE, easier to calculate MSL from that than the other way around, should be very easy to implement

mc-hamster, 2021-03-30:
I'm laughing at this because we used to use HAE but the request in #363 was to switch to MSL.

@sachaw Maybe this is a good topic to try out the discussion system you messaged me about? IMHO - We should pick one or the other.

a-f-G-U-C, 2021-03-30:
> we used to use HAE but the request in #363 was to switch to MSL

A couple of points for clarity.

1. Issue #363 was about the HAE/MSL model  inconsistency between NMEA and UBLOX readings; it made no recommendation **which** model to use, just to **choose one** and use it consistently. I had no input in the choice of MSL over HAE, it was pretty much a coin toss decision by @geeksville at the time. 
2. half a year ago, when I opened issue #363, all the software that I was using with Meshtastic was able to ingest HAE/MSL equally well, so I had no personal preference of elevation model and not even a clear understanding of the key differences. Now **I do** have a preference, which I have explained and substantiated.

I apologize for the apparent change of mind (and it's perfectly OK to laugh at me), but I think HAE is the better choice based on the currently available data. I hope this makes sense.

a-f-G-U-C, 2021-09-10:
**RELEVANT UPDATE**

A nice side effect of the recent set of changes to the GPS code is that it **removed all the stoppers** to solving this issue. At the moment, the Meshtastic device firmware can be migrated to the HAE model by simply appending `-DGPS_ALTITUDE_HAE` to the compile flags in platformio.ini; all the necessary code tweaks are already in place for the change to "just happen".
I leave the decision on this issue and the timing of the change in the capable hands of the project maintainers  :)

Thank you

geeksville, 2021-09-18:
also - @a-f-G-U-C you could add a haeAltitude field to the position protobuf.  And if only populated when GPS_ALTITUDE_HAE is supplied it would only get bigger in that case.  This would allow position consumers to work with either MSL or HAE (I think).

a-f-G-U-C, 2021-09-19:
Thanks @geeksville - I can definitely do that, and in fact I have a larger proposal in the pipeline, for several optional field additions to the position protobuf. 

However, my preferred way forward at this point would be to first **change the default altitude model to HAE**, i.e: if only one altitude field is included in a default minimal size position packet, with no further information around it to disambiguate, that field should contain the HAE altitude (instead of MSL as it does now), as the most useful and versatile of the two alternatives. That's why my proposal re-purposes the existing altitude field instead of defining a new one. 
Like @sachaw very well pointed out, it's easier (for the position consumer) to calculate MSL from HAE than the other way around. 

I'm aware it's a breaking change, and I half-expect my proposal to be rejected on this basis, but I also know it's a good change and we're at a maturity level where *we can still afford* breaking changes of this (relatively small) magnitude. Also, I note that in the 6 months this proposal has been open for discussion, there has been some positive feedback, but no strong opinion **against** it.

a-f-G-U-C, 2021-10-04:
Apologies @geeksville and all - while my statement above might make sense in a static variable scenario, it is not the **protobuf** way to handle changes; the correct solution is indeed to add a new field. Took me a while for this particular penny to drop :)

I will include this new field in the update from proposal #873 . Leaving the previous comment in place for historic reference.

closed 2021-10-24 by sachaw