meshtastic/firmware #363: GPS altitude model inconsistency between u-blox and NMEA protocols

a-f-G-U-C, 2020-09-04:
## Describe the bug

Meshtastic supports two different protocols for reading the onboard GPS: 
* NMEA protocol, more generic but less optimal
* u-blox protocol, rich and optimized, but only available on some GPS models

The GPS measured altitude can be expressed using two reference models: MSL or HAE, which I have described in issue #359 

Currently, Meshtastic reports the MSL altitude when using the NMEA protocol, and HAE when using u-blox, resulting in an uncorrectable inconsistency (since the protocol used for reading the GPS is opaque to the user). 

## Expected behavior
The same altitude model, whether HAE or MSL (or both!), should be used consistently across all GPS protocols.

## Code references

The u-blox code reads the HAE altitude:
https://github.com/meshtastic/Meshtastic-device/blob/88f8bbe21e1ca3696e92570da20422211c6c4624/src/gps/UBloxGPS.cpp#L168
> NOTE: for MSL altitude, use `getAltitudeMSL()`) 

The NMEA code reads the MSL altitude:
https://github.com/meshtastic/Meshtastic-device/blob/88f8bbe21e1ca3696e92570da20422211c6c4624/src/gps/NEMAGPS.cpp#L49
> NOTE: for HAE altitude, use `reader.altitude.meters() + read.geoidHeight.meters()`)

## Additional info
This is related to, but a different (and more urgent) issue than #359.

geeksville, 2020-09-08:
thanks for the informative writeup!  fixing!

a-f-G-U-C, 2020-09-09:
Glad I could help, closing with thanks.

commit c0073025 (2020-09-11): fix #363 gps altitude - based on tip by @a-f-G-U-C
closed 2020-09-09 by a-f-G-U-C