meshtastic/firmware #372: Feature Ideeas

cezarlacatus, 2020-09-09:
Hello, I have 2 normal TTGO 2.0 boards.I just started using meshtastic and I would have some ideeas to make this project better:

1.The use of Wifi: Ok, almost every ttgo board has wifi modules onboard,so why we are not using them? 
I am thinking about this implementation using 4 Boards paired in the same group:
Board 1 send a message throught LORA to Board 2.Board 2 is connected throught wifi to a server that is hosted on a simple VPS(Or we can have a secured comunity server).Board 3, and Board 4 is not in LORA range with Board 1 or 2, but Board 3 has wifi access.Board 3 will download the message from the server and broadcast it with LORA to Board 4.

In this way, users can communicate even if LORA is not sufficient.You can comunicate and connect with nodes that are on the other side of the earth throught this way.Moreover, the user can specify multiple wifi AP and the board will auto connect from time to time to upload and download messages.

2.Implement repeators: Lets say you want to extend the lora range.It would be great if we could have an open source Repeater code that we can use on boards without OLED display(ex. LoRa32u4 II) in order to extend the lora mesh network.

In my opinion the wifi ideea is the most important one as the boards are equiped with this module and will come handy for some users(I would find this usefull to send a message to my little brother from another city).We should use the internet to our advantage to drive this project with extended founctionality.

Thank you and hope my ideeas would be considered.

geeksville, 2020-09-10:
oh yes - repeater mode is very soon in the queue.  As is wifi.  thanks - you also might dig the meshtastic.discourse.group where we are talking about this a bit.

syssytheboss, 2020-09-10:
@geeksville Thank you verry much.I don't actually know where is the discussion group.I think you refeer to the forum.It would be super cool if we can also build a discord server to facilitate the communication.

Anyway, thank you for taking my ideeas in mind 😊

mc-hamster, 2021-03-27:
Closing this request. Both #1 and #2 are done.

closed 2021-03-27 by mc-hamster