meshtastic/firmware #373: TBeam shows "No GPS" in the upper right corner after some random time.

drewsed, 2020-09-09:
TBeam v1.1 sx1262 neo8m shows "No GPS" after some random time.
Sometimes this happens after a fresh boot, but most of the time a reset is fixing the issue for another random time.

My Setup is a mesh with 2/4 devices:
- 2 TBeams (online)
- 2 Heltecs (offline)

I was trying to capture some debug output, but as always it does not occour. 
**BUT** I captured the debug output having this issue on a fresh boot:
``
No GPS from Boot:

Emitting reboot packet for serial shell
��␀␂H␁booted, wake cause 0 (boot count 1), reset_reason=reset
I2C device found at address 0x34
axp192 PMU found
I2C device found at address 0x3c
ssd1306 display found
done
Meshtastic swver=0.9.5, hwver=1.0-EU865
Setting random seed 1199878518
Total heap: 265492
Free heap: 238292
Total PSRAM: 4194252
Free PSRAM: 4194252
NVS: UsedEntries 111, FreeEntries 519, AllEntries 630
chip id detect 0x3
Detect CHIP :AXP192
OUTPUT Register 0x5f
AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
----------------------------------------
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
SRC REG:0xcf
Charging enable is enable
Charging target-voltage : 0x2
 end when the charge current is lower than 10% of the set value
Charge current : 1000.00 mA
enable [0x40]val:0xdc
enable [0x41]val:0xff
enable [0x42]val:0x3b
Turning on screen
Read RTC time as 0 (cur millis 143) valid=0

ERROR: No UBLOX GPS found
Hoping that NEMA might work

RadioConfig reset!
Installing AES128 key!
Initial packet id 118979972, numPacketId 4294967295
Loading saved preferences
Loaded saved preferences version 11
Installing AES256 key!
NODENUM=0xb21bb5b0, dbsize=4
Starting meshradio init...
Set radio: name=XXXXXXXXXXXXXX, config=3, ch=3, power=17
RF95 init result -2
Warning: Failed to find RF95 radio
Starting meshradio init...
Set radio: name=XXXXXXXXXXXXXX, config=3, ch=3, power=17
SX1262 init result 0
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Battery 4170mV 100%
Screen: Started...
sending owner !ac67b21bb5b0/Julia/J
Update DB node 0xb21bb5b0, rx_time=0
old user !ac67b21bb5b0/Julia/J
updating changed=0 user !ac67b21bb5b0/Julia/J
Adding packet record (id=0x07177d87 Fr0xb0 To0xff, WantAck0, HopLim3 Payload:User)
enqueuing for send (id=0x07177d87 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
txGood=0,rxGood=0,rxBad=0
Starting low level send (id=0x07177d87 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
showing standard frames
Transition powerFSM transition=boot timeout, from=BOOT to=ON
Setting bluetooth enable=1
Pre BT: 223076 heap size
Starting bluetooth
BLE task running
registered service 0x1800 with handle=1
registered service 0x1801 with handle=6
registered service 6ba1b218-15a8-461f-9fa8-5dcae273eafd with handle=10
registered service cb0b9a0b-a84c-4c0d-bdbb-442e3144ee30 with handle=18
BLE advertisting type=0, Private=0, Device Address: 39:ac:67:b2:1b:b5

Completed sending (id=0x07177d87 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
Lora RX (id=0x07177d87 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=5.25)
Rx someone rebroadcasting for us (id=0x07177d87 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=5.25)    
Found existing packet record for fr=0xb21bb5b0,to=0xffffffff,id=118979975
Ignoring incoming msg, because we've already seen it: fr=0xb21bb5b0,to=0xffffffff,id=118979975,hop_limit=2
Lora RX (id=0x0810a698 Fr0x18 To0xb0, WantAck0, HopLim3 encrypted rxSNR=5.25)
Adding packet record (id=0x0810a698 Fr0x18 To0xb0, WantAck0, HopLim3 encrypted rxSNR=5.25)   
FIXME not implementedFIXME-update-db Sniffing packet
Delivering rx packet (id=0x0810a698 Fr0x18 To0xb0, WantAck0, HopLim3 Payload:User rxSNR=5.25)
Trigger powerFSM 3
Ignoring incoming packet - not a position
Forwarding to phone (id=0x0810a698 Fr0x18 To0xb0, WantAck0, HopLim3 Payload:User rxSNR=5.25)
Update DB node 0xb21bb718, rx_time=0
old user !ac67b21bb718/Andi/A
updating changed=0 user !ac67b21bb718/Andi/A
Telling client we have new packets 1
Telling client we have new packets 1
No BLE notify
Sending position to mesh
Update DB node 0xb21bb5b0, rx_time=0
Node status update: 4 online, 4 total
showing standard frames
Stripping time 0 from position send
Adding packet record (id=0x07177d88 Fr0xb0 To0xff, WantAck0, HopLim3 Payload:Position)
enqueuing for send (id=0x07177d88 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
txGood=1,rxGood=2,rxBad=0
Starting low level send (id=0x07177d88 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
Completed sending (id=0x07177d88 Fr0xb0 To0xff, WantAck0, HopLim3 encrypted)
Trigger powerFSM 1
Transition powerFSM transition=Press, from=ON to=ON
Setting fast framerate
Setting idle framerate
Trigger powerFSM 1
Transition powerFSM transition=Press, from=ON to=ON
Setting fast framerate
Setting idle framerate
Lora RX (id=0x07177d88 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=4.25)
Rx someone rebroadcasting for us (id=0x07177d88 Fr0xb0 To0xff, WantAck0, HopLim2 encrypted rxSNR=4.25)
Found existing packet record for fr=0xb21bb5b0,to=0xffffffff,id=118979976
Ignoring incoming msg, because we've already seen it: fr=0xb21bb5b0,to=0xffffffff,id=118979976,hop_limit=2
Trigger powerFSM 1
Transition powerFSM transition=Press, from=ON to=ON
Setting fast framerate
Setting idle framerate
Trigger powerFSM 1
Transition powerFSM transition=Press, from=ON to=ON
Setting fast framerate
Setting idle framerate
Update DB node 0xb21bb5b0, rx_time=0
Node status update: 4 online, 4 total
Update DB node 0xb21bb5b0, rx_time=0
Node status update: 4 online, 4 total
Update DB node 0xb21bb5b0, rx_time=0
Node status update: 4 online, 4 total
``

I will update this post, if the error occurs during the evening or night, lets see.
I also thought about not getting the issue while plugged into a 5V charger.

My second device running on 18650 cell had the issue twice during the investigation ^^

geeksville, 2020-09-09:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/timed-reboot-feature/1226/5

drewsed, 2020-09-10:
Hmm, still no "No GPS" after running for about 14 hours.

drewsed, 2020-09-13:
"Ni GPS" never appeared for me again since 0.9.6 and 0.9.7.

I will close this.

closed 2020-09-13 by drewsed