meshtastic/firmware #374: Implement HTTP server

mc-hamster, 2020-09-10:
Features:

- Utilizes the esp32 web server library
- Feature toggle so this can be turned on/off from configuration.h
- Very limited statistics

Initial assumption is this will be USB powered. No power savings will be implemented.

More specific features will come in future tickets.

This is only to get the framework of being able to serve HTTP.

@geeksville Please assign this to me.

geeksville, 2020-09-10:
I think this is a good idea.  With one caveat/opinion - we should only serve up static HTTP goo necessary to provide all the services either client side or client side talking to our existing meshtastic API service on port 4403.

Or perhaps: we could provide a small surface area REST API which just accepted a PUT to /ourresetapi/toradio and also a streaming GET reply of a series of fromradio protobufs.  We could provide those two REST endpoints with either bare protobuf framing or as JSON (but JSON based on the automatic JSON protobuf code).  This would allow us to keep essentially 'one' API across bluetooth, USB, TCP and HTTP.

syssytheboss, 2020-09-10:
@mc-hamster It would be cool if the HTTP server could be accessed also by non-lora devices.I have a dummy TTGO T-Display that is only equiped with Wifi(I think it may have BT),so it would be cool to use an API(implemented by the server) and get messages displayed.

As a suggestion pure nodejs with express would be the best suit for this project.

For this moment I would really like to see a demo working system that use wifi and that can connect to the server for querring and placing messages.

Later on you could implement autentification and even communication chanels inside that specific server.

mc-hamster, 2020-09-13:
See PR #384

closed 2020-09-13 by mc-hamster