meshtastic/firmware #375: Implement WIFI_MODE_STA

mc-hamster, 2020-09-10:
Features:

- Toggle in configuration.h to enable wifi mode
- Configure SSID and Password from configuration.h
- Roam / Reconnect to AP when disconnected from current AP and connect to another AP
- Don't hang the device if unable to connect to an AP
- Display Network status on display

Assumptions:
- This will be a bit more than enough to get started with implementing various services over IP.
- AP mode will be implemented at a later date.
- For the initial implementation, we will assume that the device will be powered by USB and we won't deal with power management.
- Initial target is to support reliable connections for a TCP/HTTP server
- It may be useful to support other protocols in the future. 

@geeksville Please assign this to me.

syssytheboss, 2020-09-10:
@mc-hamster Looking forward to your ideeas.

mc-hamster, 2020-09-13:
See PR #384

closed 2020-09-13 by mc-hamster