meshtastic/firmware #381: User interface to enter the text message?

wzab, 2020-09-12:
Does meshtastic support any user interface allowing the user to enter the text message?
I would like to propose two possibilities:

1. Using the Morse key (for example built from a capacitive sensor like that one: https://pl.aliexpress.com/item/32671645328.html ) and Morse code decoder
2. Using a simple capacitive keyboard (e.g., https://pl.aliexpress.com/item/4000001770650.html ) where the character may be selected by two consecutive key presses. The first key-press selects the group of characters displayed in the lowest part of the display, the second key-press selects the individual character.

With such a feature it could be a perfect standalone communication device.

retrodime, 2020-09-12:
+ Ideea:
- Some kind of BLE keyboard
 - Or buttons like the old phones

Anyway, this require additional soldering and custom code.....but it would be cool to come with an ideea

benner, 2020-09-13:
There is also I2C keyboards (e.g. https://www.aliexpress.com/item/32963872643.html?spm=a2g0o.productlist.0.0.71b34486dvZNt4&algo_pvid=c784ffa4-7a9b-4c64-a27b-6a48c8b80888&algo_expid=c784ffa4-7a9b-4c64-a27b-6a48c8b80888-0&btsid=0bb0620316000065546922971ee1f5&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_ ) that should not hard to support too

wzab, 2020-09-24:
I have tried to adapt the Morse decoder by David Bird - https://github.com/G6EJD/ESP32-Morse-Decoder and tested in on Bluepill - https://gitlab.com/WZab/wzab_arduino_projects/-/tree/master/morse_decoder . The results were however unsatisfactory due to problem of detection of the transmission speed.
Then I have decided to adapt the [Morse decoder](https://github.com/oe1wkl/Morserino-32/blob/master/Software/src/morse_3_v3.0/MorseDecoder.cpp) used by Willi Kraml in his wonderful [Morserino32](https://github.com/oe1wkl/Morserino-32). The results of tests on Bluepill are quite good. The simplified decoder is available [here](https://gitlab.com/WZab/wzab_arduino_projects/-/tree/master/BP_morse_decoder). I can try to integrate it with Meshtastic on TTGO T-Beam in reasonable future (but not immediately).

m0, 2021-01-01:
This seems to be a duplicate of https://github.com/meshtastic/Meshtastic-device/issues/62 or did I miss something?

geeksville, 2021-01-03:
thanks @m0 yep - I agree.  
@mzab if/when you can do this as a #ifdef HAS_MORSE or somesuch I'd eagerly merge.
Closing to keep all discussion in #62

closed 2021-01-03 by geeksville