meshtastic/firmware #383: Flag in header > Forward raw packet.

bobricius, 2020-09-12:
Is possible add flag bit in to header "do not decode payload just forward" ?
It can help for some stupid simple devices like ARMACHAT or temperature sensors just use meshtastic network nodes
for communication without complex implementation of all protobuf and encryption things.

my idea is use Meshtastic units as standalone routes ... is it possible?

syssytheboss, 2020-09-13:
I think @geeksville will add this as it is a part of something that may be a relay system, also it would be cool to sign the message with a list of devices ID in order not enter in a repeat loop of the group.
Anyway every cool feature that meshtastic plans to implement, comes after the V1.0 Release as he needs to get rid of some bugs.

Cant wait to see the meshtastic in 2-3 months.I think I'm gonna order some modules from now.

geeksville, 2020-09-13:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/armachat-standalone-offgrid-messenger/1242/9

bobricius, 2020-09-17:
is possible implement raw packet forward ? https://github.com/meshtastic/Meshtastic-device/issues/383
I think it is only few code lines.

geeksville, 2020-09-17:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/the-1-1-feature-work-list-triage-thread-lobby-for-your-features-here/1324/4

mc-hamster, 2021-03-27:
Marking closed.

The nodes will relay packets that are on the same frequency and spreading factor provided it has some of our basic packet headers. It will do so without decoding or decrypting the payloads.

We need a basic meshtastic header to ensure that we keep track of the message id and hop count so we don't end up flooding the air with replayed packets.

closed 2021-03-27 by mc-hamster

bobricius, 2021-03-28:
it is already implemented and working ?

mc-hamster, 2021-03-29:
Yup. 

https://github.com/meshtastic/Meshtastic-device/blob/69de2a047fa34dee80dd32aeb23d021089d4fa2f/src/mesh/RadioLibInterface.cpp

Line 270