meshtastic/firmware #386: When the ble radio powers down, the wifi radio also powers down

mc-hamster, 2020-09-13:
When the ble radio powers down, the wifi radio also powers down.

When the ble radio comes back up, the wifi radio stays down.

a-f-G-U-C, 2020-09-13:
As you probably know, ESP32 has only one ISM band transceiver and one antenna, shared between the WiFi and BLE stacks through some clever software tricks, which create some interesting corner cases.

What's important to keep in mind when debugging this issue is that you don't power up/down the "BLE radio" or the "WiFi radio", but simply **THE radio**. Any discrepancy that you encounter is likely a software stack issue.

mc-hamster, 2020-09-13:
> As you probably know, ESP32 has only one ISM band transceiver and one antenna, shared between the WiFi and BLE stacks through some clever software tricks, which create some interesting corner cases.

Ah ha! I didn't know it was some kind of soft radio with one transceiver. That makes total sense.

> What's important to keep in mind when debugging this issue is that you don't power up/down the "BLE radio" or the "WiFi radio", but simply **THE radio**. Any discrepancy that you encounter is likely a software stack issue.

This helps with debugging. Thanks mate!

mc-hamster, 2020-09-13:
Yes, I found the problem.

I took initWifi(); out of BluetoothUtil.cpp because I needed to more control over that along with decoupling it from bluetooth.

What I need to do is to create a new method (maybe: reinitWifi()) that's a little lighter weight than the full initialization.

I'll make the decision on how to fix after I finish refactoring to satisfy Kevin's comments in my other PR.

mc-hamster, 2020-09-14:
This is fixed in my branch.

closed 2020-09-14 by mc-hamster

mc-hamster, 2020-09-19:
The fix really didn't fix anything. There's a bug in the SDK.