meshtastic/firmware #392: RFE: include GPS accuracy values

a-f-G-U-C, 2020-09-15:
## Background
In popular belief, a GPS position is naively thought as a POINT in a 3-dimensional space, defined by a triplet of values: latitude, longitude and altitude. This is unfortunately incorrect and tends to create unrealistic expectations. 

In reality, a more accurate model for the GPS position would be a vertical CYLINDER, centered around a 3D lat/lon/alt point, with a horizontal radius and vertical height derived from the HDOP/VDOP readings of the GPS receiver. The actual position is assumed to be SOMEWHERE inside this cylinder.

And then there’s the time dimension: the moment when a data point was measured is usually different to the moment when the point is transmitted, to when it’s received by the other end, and finally to when it’s plotted on the map. 

Depending on the application, there are practical limits to the usefulness of a GPS position. For example, a horizontal uncertainty of 50m may be OK for recreational boating, but not for auto-landing a UAS; an uncertainty of 500km is useless for most practical purposes. Likewise, a 6-hour old position is OK for geocaching, but not for paragliding. And so on. 

In conclusion: there are, at the very minimum, six values that define a GPS position properly: latitude, longitude, altitude, HDOP, VDOP and timestamp. All these values are available from GPS receivers as a matter of standard operation. 

## Currently
Meshtastic reads all necessary values from the GPS, however it only transmits over the air the lat/lon/alt triplet without the DOP values and timestamp. This results in potentially invalid location information being transmitted to peers, while taking away their ability to discard invalid or useless data points. This is often worse than no data at all. 

**Relevant code section**
https://github.com/meshtastic/Meshtastic-device/blob/cc6b50002960548457dcf0ff7f44b0fc4ff742ca/src/mesh/mesh.pb.h#L88-L94

## Request
It would be great if the Meshtastic mesh `Position` object could be updated to include all the 6 values that determine the GPS position, in order to enable client applications to properly classify, ingest or discard data points based on their individual accuracy requirements. 

Also, fixes that are obviously invalid (for example, HDOP=9999 etc) should probably be treated as **non-fix** and discarded at source. 

**Additionally:**
* HAE altitude is another strong candidate for protocol inclusion (either as absolute value or as geoidal height), per issue #359
* SIV (satellites in view) and “fix type” (2D/3D/etc) are also somewhat *interesting* metrics (though not quite *actionable*) that could be included on an optional basis.

Thank you

geeksville, 2020-09-15:
good idea, though possibly best to only support this feature on the local node.  Because in the long range setting adding even a few bytes to the position packet (for things like timestamp, hdop, vdop) will take noticeably more airtime.

a-f-G-U-C, 2020-09-16:
Fair point, never thought of it in these terms, but obviously every new data field has a cost in the bandwidth budget.

Perhaps it might make sense to define two different `Position` objects, say "Lite" and "Full", selected at runtime based on the application's accuracy requirements vs power/bandwidth budget. 

Anyway, this looks fairly complex; I might break it down into smaller issues and take them one at a time. :)

DylanHoen, 2020-09-24:
I just realized that I created an issue that was very similar to this one:
https://github.com/meshtastic/Meshtastic-device/issues/432
Mine differed in keeping of the latest coordinate and a second latest accurate coordinate, and in explaining how they could be drawn to the map.

I certainly second the issue of older, or inaccurate, coordinates overwriting good ones, being a bug if hikers are depending on it for safety.
If bandwidth is an issue, maybe there could be a prioritize GPS accuracy over responsiveness button in the app.

DylanHoen, 2020-10-08:
You could add a basic accurasy and time stamp in just 1 byte.  4 bits, 0 to 15.  That is how many times to left shift 1 to make a distance accuracy sphere.  That could be 1 meter to 32.768km+.  That could be 1 second ago to 9 hours, 6 minutes, and 8 seconds+ ago.
For time, you usually need a lot of precision, going back to the 1970's or 1900 for 0, but if you measure how long ago something happened, you don't need as many digits as a time stamp.  Gps coordinates usually come with an accurate date stamp.  That can exist in ram without transmitting it.  When transmittjng, just transmit which power of 2 seconds ago it was recorded.  The receiver can use that power of 2 and its internal real time clock to estimate the date stamp.  If it retransmits, it subtracts the date stamp from its current real time clock time, and transmits that power of 2.  If the real time clock is forced to change, either some smarts can be used to addjust the date stamp to the new time, or the date stamp can be changed to the maximum time, and transmitted as 15 (2^15).  If the real time clock looses power, it becomes 15 on bootup.

mc-hamster, 2021-03-27:
Please see https://github.com/meshtastic/Meshtastic-device/issues/432 for more information around this feature request.

I'm going to close #432 as a duplicate of this ticket.

closed 2021-10-24 by sachaw