meshtastic/firmware #396: Heltec acts as though button is held down while connected via USB to computer running python api

timgunter, 2020-09-16:
1) Connect heltect lora 32(firmware v1.0.0) to computer via USB
2) Run "python -m meshtastic --reply" (v1.0.8)
3) Observe device acts as though button is held down and goes into screen brightness adjustment mode forever
4) Pushing button does not exit screen adjustment mode

When I connect a T-Beam (firmware v1.0.0), this does not happen.

geeksville, 2020-09-16:
ooh - this was probably an unfortunate bug of fixing RTS handling on OSX (which was applied on all platforms).  I'm guessing that that board (which I don't personally have alas) also hooks RTS to their button input.

Hmm - I guess I'll have to add some code on the device side to disable that 'button' entirely if the API is in use...

mc-hamster, 2020-09-16:
The RTS change on the API may have had some unintended consequences and be the source of the problems I was seeing yesterday.

I've spun up a linux system and will try rolling back that one change in that environment.

If it works there, let's reopen the OS X bug and roll back the RTS change.

geeksville, 2020-09-16:
@mc-hamster yep that was it - see my comment on the bug I just closed and pull the latest python tool.

geeksville, 2020-09-16:
btw - 1.0.10 on pypi should make the button work nicely again on Linux and Windows.  Though it will still be messed up on os-x.

mc-hamster, 2021-03-27:
Closing this ticket. This behavior no longer happens.

closed 2021-03-27 by mc-hamster