meshtastic/firmware #407: dev-wifi: wifi does come back up after power down

mc-hamster, 2020-09-18:
WiFi doesn't come back up after being shut down.

This is due to a bug in the Espressif SDK was introduced in October 2019. Here's the PR to fix the issue in the SDK.

Seems like this bug was introduced when the wifi long range feature was rolled out and it's unknown how this fix will affect the long range feature.

espressif/arduino-esp32#4114

mc-hamster, 2020-09-19:
The error message I was getting that lead me to the PR from espressif:

[WiFiSTA.cpp:158] begin(): disconnect failed!

geeksville, 2020-09-19:
@mc-hamster that fix looks good to me.  I'll merge it into our local fork of arduino-esp32 this weekend.

mc-hamster, 2020-10-22:
@geeksville I commented on the upstream issue and it seems like they've merged it into espressf:master a few days ago! We should update our fork.

mc-hamster, 2020-10-22:
I'm going to take assignment on this.

mc-hamster, 2020-10-22:
I'm chickening out of that. Just did a diff between the two -- there's a ton of changes on both sides on the ble area. I wouldn't be able to confidently merge the two.

mc-hamster, 2020-12-23:
@crossan007 Thanks for your feedback of the http server not coming back up. This is now fixed too.

closed 2020-12-23 by mc-hamster