meshtastic/firmware #413: Create a basic HTML chat client

mc-hamster, 2020-09-19:
Create a basic HTML chat client.

- Chat history available as a json object
- Post message json endpoint
- XMLHttpRequest to auto refresh the chat window
- Very basic HTML user interface

thepoweroftwo, 2020-09-25:
Could be relevant to this project: 
We have just published a meshtastic javascript library (https://github.com/thepoweroftwo/meshtastic.js)
Right now it only supports connecting to a device via bluetooth.
From a quick look i haven't found much about how dev-wifi is implemented. Maybe a connect-via-wifi interface could be added to meshtastic.js too?

mc-hamster, 2020-09-25:
@thepoweroftwo That's really cool!

There are two APIs being exposed over the wifi interface. @geeksville is working on exposing the protobuf API over TCP 4403 while I'm working on a highly abstracted API with JSON endpoints. Both of these are not yet done.

geeksville, 2020-09-27:
@thepoweroftwo wow that is amazeballs.  Are you planning on build a full JS client based on this lib?  Super nicely done!

For access to the devices over wifi the API will probably end up looking like this: https://github.com/meshtastic/Meshtastic-device/wiki/HTTP-REST-API-discussion

thepoweroftwo, 2020-09-27:
Actually yes - but I still haven't decided on the scope and details. 
What I'm currently thinking about is using a framework like vue.js for the app and, since I'm doing UX design on my main job, creating a nice user interface for it. 
Maybe this - once ready - could also be used as an interface design for the app @mc-hamster is creating? Also, if anyone has ideas about my thoughts, feel free to share. 

Thanks for the http api link, it looks very exciting!

mc-hamster, 2020-09-27:
@geeksville Proposed that we expose the protobufs over rest. I think it would then be possible to have a web interface use either one depending on device support. BLE does use a lot less power than WiFi. I was unaware that BLE worked on iOS devices from within the browser?

David M has put together the following UI for us. It's currently being served by the on board web server. It's just waiting for the endpoints to receive messages and to send messages, but it's otherwise fairly complete:

[image]

thepoweroftwo, 2020-09-28:
This already looks usable. I wonder if it is displayed on the phone like that? Seems a bit small (chat area could be stretched to use the whole screen space and left channel bar could be a fly-in from the left). Otherwise really cool!

Regarding BLE on Safari: There's the catch. Apple decided against implementing features like the web bluetooth api and web serial. https://www.zdnet.com/article/apple-declined-to-implement-16-web-apis-in-safari-due-to-privacy-concerns/
So web bluetooth currently does not work on desktop and mobile safari.

luandro, 2020-09-28:
This is really exciting, great job, looking forward to trying it out :)

mc-hamster, 2020-09-28:
Right now, David's main effort for the WebUI has been to get the basic layout going. I'm sure it'll be made more responsive later 👍 

He's done a bang-up job on the design. It's just a few DIVs and embedded fonts for some graphics.

mc-hamster, 2020-10-01:
@thepoweroftwo David (he made the UI above) and I were looking at your API had had a thought.

What do you think about having that API interface with the protobufs under both BLE and HTTP/REST and then just a toggle to go between the two. You have a lot of plumbing already there and it would be a real service if there were a single well used javascript API that interfaces over both ble and http, instead of having multiple.

thepoweroftwo, 2020-10-01:
The API can be easily extended by adding another class. Like the current IBLEConnection there could be an IHTTPConnection. Both extend the parent class IMeshDevice which provides the API functions (send, receive events, etc).

Is that what you were thinking about? Or do you mean making the connection layer intransparent (abstracted) to the API user?

Another thought regarding the API: Right now I am using the protobuf objects as an internal data format. So when querying the node db you get back a protobuf NodeList object, when receiving a message event you get back a FromRadio object and so on. This is fine as long as protobufs are used for the transport. The ideal way would be abstracting the protobuf layer, that way the JS API would be also independant from the transport layer being used to send/receive the data from/to device. Not necessary, just to keep in mind.

mc-hamster, 2020-10-02:
I think another class would be fine.

I'm busy with my day job this week and weekend, but I'll have time next week to look at exposing the protobufs over HTTP.

Are you on our slack?

thepoweroftwo, 2020-10-02:
Ok. I've just joined the slack team

Noki, 2020-11-01:
This sounds cool. If not already done this could be transformed into an installable PWA with only a few lines of code, allowing users to "Install" it on their phones or desktop pcs and later access it like a normal app: https://www.youtube.com/watch?v=LWRdBywm4Zo

mc-hamster, 2020-12-22:
Assigning this to @crossan007 now that he's part of the org.

crossan007, 2020-12-22:
👀 

I didn't realize there was already work-in-progress when I started https://github.com/crossan007/meshtastic-web

Hopefully not stepping on toes or causing much rework :-)

mc-hamster, 2020-12-22:
The only thing that used to work was a proof of concept with  html text area, an input field and a submit button with a form tag wrapping it all together.

mc-hamster, 2021-03-27:
Closing this ticket. We have it in the Meshtastic-web repository.

closed 2021-03-27 by mc-hamster