meshtastic/firmware #415: remove support for 8 bit node id/packet nums

geeksville, 2020-09-19:
(now that 1.0 is out the old original system can be deprecated)

sachaw, 2021-12-29:
@mc-hamster is this still applicable?

mc-hamster, 2022-03-10:
generatePacketId(); // FIXME - ugly way to init current_packet_id;
/// Generate a unique packet id
// FIXME, move this someplace better
PacketId generatePacketId()
{
    static uint32_t i; // Note: trying to keep this in noinit didn't help for working across reboots
    static bool didInit = false;

    assert(sizeof(PacketId) == 4 || sizeof(PacketId) == 1);                // only supported values
    uint32_t numPacketId = sizeof(PacketId) == 1 ? UINT8_MAX : UINT32_MAX; // 0 is consider invalid

    if (!didInit) {
        didInit = true;

        // pick a random initial sequence number at boot (to prevent repeated reboots always starting at 0)
        // Note: we mask the high order bit to ensure that we never pass a 'negative' number to random
        i = random(numPacketId & 0x7fffffff);
        DEBUG_MSG("Initial packet id %u, numPacketId %u\n", i, numPacketId);
    }

    i++;
    PacketId id = (i % numPacketId) + 1; // return number between 1 and numPacketId (ie - never zero)
    return id;
}

thebentern, 2022-04-26:
@mc-hamster is this still an active issue?

garthvh, 2022-06-14:
Can we delete this now that 1.3 is the master branch?

garthvh, 2022-08-05:
Code is totally gone, closing

closed 2022-08-05 by garthvh