meshtastic/firmware #420: Wifi in station mode sometimes enters loops of repeatedly joining...

geeksville, 2020-09-19:
seen about 50% of boots:

```
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
JOINING WIFI: ssid=geeksville
Started Joining WIFI
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
JOINING WIFI: ssid=geeksville
Started Joining WIFI
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
JOINING WIFI: ssid=geeksville
Started Joining WIFI
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
JOINING WIFI: ssid=geeksville
Started Joining WIFI
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
JOINING WIFI: ssid=geeksville
Started Joining WIFI
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
JOINING WIFI: ssid=geeksville
Started Joining WIFI
************ [WiFi-event] event: 5 ************
Disconnected from WiFi access point
JOINING WIFI: ssid=geeksville
```

I'll bet it has to do with both BLE and wifi being connected at once and the final fix will be to "if configured for wifi, do not start BLE"

cc @mc-hamster

mc-hamster, 2020-09-19:
Few possibilities for this:

1) Wifi ssid / password combination was entered wrong
2) Device is just outside the range of the AP and is having trouble maintaining an established connection.

mc-hamster, 2020-09-19:
I've been able to reproduce this but for maybe only 10% of reboots. 

I think there's a race condition somewhere.

mc-hamster, 2020-09-19:
Fixed it.

If it fails to associate the first time, it immediate calls the disconnect event and tries to reinit the wifi stack. Within the initwifi, I was attaching the the event handler.

WiFi.onEvent(WiFiEvent);

It's ok for that to be called multiple times, but not always. Very odd.

I refactored the initwifi function to create reconnectWiFi() for that situation.

mc-hamster, 2020-09-19:
Also added a RSSI display on the screen which helped me debug this.

mc-hamster, 2020-09-20:
Fixed. Checked into dev-wifi

closed 2020-09-20 by mc-hamster