meshtastic/firmware #427: Lora Mesh Frequency Band Regulation

mightChamp, 2020-09-22:
Hello,
   As LoraWAN manage to use of frequency band in 1% of duty cycle, but I think that lora mesh can't manage this, as a node also have to transmit for neighbors, too, and 1% duty cycle is on use of frequency band.

Then what about the future of Lora Mesh if it breaks regulations.

This is an important issue for me, as I am thinking of making some product based on this. So please discuss about it regulation.

Thank You

slavino, 2020-09-22:
You can have a look on a variety of products that use LoRa technology and are certified as well (and not use LoRaWAN).

https://www.dogtrace.com/en dog tracking GPS collar as an example - uses 868MHz LoRa. I believe on their page they share also the certification, if not, just ask them...

BishopPeter1, 2020-09-24:
It depends from what region you are. In EU are frequencies (channels) where you can have 10% duty cycle...

geeksville, 2020-09-27:
> This is an important issue for me, as I am thinking of making some product based on this. So please discuss about it regulation.

I think some serious attention to regulatory requirements is a good idea.  We'd happily accept any PRs you make (and eventually I work on this if no one has contributed it first).  We should put in code to ensure that we don't accidentally violate duty-cycle limits in certain regions.

closed 2022-02-02 by garthvh