meshtastic/firmware #432: GPS accuracy circle and time stamp on map

DylanHoen, 2020-09-23:
One of my T-Beams, with likely the latest or second latest firmware and android app (September 20'th, maybe), appeared on the map 100km away, while they were all within 10 meters of me.  That GPS coordinate should have come with some kind of confidence circle and an accurate time stamp.
There is a list of different circles here:
https://en.wikipedia.org/wiki/Circular_error_probable
It seems that there could be a radius of 50% probability, but if you double it, it becomes 95% probability.  The 95% might be the circle I want to see on the map.
Each person could be drawn on the map with a shaded circle of confidence drawn around them.  Each shaded circle of confidence could get darker where they overlap (if that helps distinguish who it belongs to).
Each name on the map should have the timestamp of the GPS reading, or a measure of how long ago the the GPS reading was made, on the line below the name.
When the GPS coordinates are relayed in, through multiple mesh paths, the time stamps should be used to keep the latest one, and maybe a second coordinate, which is the latest coordinate that's confidence was within a certain radius, like 10 meters.  If the latest measurement is less accurate than the 10 meter accuracy measurement, Both coordinates could be drawn on the map, with the older one being greyed out.  Measurements that are older than an hour ago, or a day ago could be greyed out fainter and fainter.

DylanHoen, 2020-09-24:
I just realized that this issue was very similar:
https://github.com/meshtastic/Meshtastic-device/issues/392
Mine differed in keeping of the latest coordinate and a second latest accurate coordinate, and in explaining how they could be drawn to the map.

a-f-G-U-C, 2020-09-30:
Sounds interesting, but isn't this a job for the phone app? The app runs on a high powered CPU with lots of RAM which can easily keep a backlog of data points and perform statistic calculations in real time.

DylanHoen, 2020-10-01:
I had multiple nodes in different parts of my buildong, and would come in and out of bluetooth range with them, and connect with them.  I would get a different batch of nodes on the map each time.  Some of them popped on the map with a name that they hadn't had in days.  I assumed that the nodes were storing information and forwarding it later, like if a hiker hiked from one group of people to another, the first group's messages would be relayed to the second group.  I then found a wishlist for a future version that had something similar on it, so I guess it isn't implimented yet.  I guess the nodes forward messages as soon as they are received, and only store them for the bluetooth connection to the cell phone.  Anyways, a radius of certanty, and a timestamp of gps coordinate acquisition could help the cell phone manage coordinates that come in from multiople bluetooth connections, if the cell phone continues to store gps coordinates between bluetooth connections to different nodes.