meshtastic/firmware #437: Display multiple GPS coordinate formats

Kak1234, 2020-09-25:
It would be nice to have the option to see the GPS coordinates in the Meshtastic radio screen in different formats that are compatible with topographic maps.

The suggested formats in **WGS84 geodetic datum**:

**LatLong in Decimal degrees** (not usable with paper maps but good for computers & apps) (ie **48.858222, 2.294500**)
**UTM** with space between easting and northing (usable with maps and most navigators are familiar with) (ie **31U 448252 5411935**)
**MGRS** / USNG [truncated to 10m](https://www.maptools.com/tutorials/mgrs/quick_guide) and with a space between easting and northing (very readable, easy to learn, and easy to use with paper maps, NATO and Search and Rescue standard) (ie **31U DQ 4825 1193**)

I'm suggesting this formats because they are pretty international and standard in the professional and recreation world.

The coordinate formats could be chosen through the API or phone app.

aHVzY2g, 2020-10-23:
Just for fun, and as it's kind of useful. I would love to see "plus code"/OLC
https://en.m.wikipedia.org/wiki/Open_Location_Code

geeksville, 2020-10-30:
@aHVzY2g thanks for mentioning plus codes, I had never heard of them and it is nice to learn a new thing.

mc-hamster, 2021-01-01:
DMS format would require 37 characters. In the current layout, we have 22 characters to work with. Implementing this would require redesigning the screen with the gps information.

Example worse case:

122° 6' 25.7832'' S 122° 6' 25.7832'' W

We can reduce the this by slightly changing the spacing, but that won't get close to 22 characters.

sachaw, 2021-01-01:
+1 for MGRS

mc-hamster, 2021-01-01:
Hmm ... Maybe the altitude can be hidden when longer coordinate formats are being displayed.

sticilface, 2021-02-14:
I think this would be a great addition.  I would also appreciate if the UK grid reference system, or indeed others were possible.  An OS grid reference would allow you to find someone in the UK on a paper map which would be quite useful.  Especially in an emergency with no nearby working phone!  

I also think displaying the position of the other node, maybe with a magnetic bearing so you can use your compass to find them would be a good idea.

tomasbrincil, 2021-05-13:
+1 for MGRS

mc-hamster, 2022-01-17:
closed.

this is done.

closed 2022-01-17 by mc-hamster