meshtastic/firmware #451: Public key based authentication and addressing

wzab, 2020-09-30:
To enable "bullet-proof" privacy, it would be good to provide an option of using assymetric cryptography both for protecting the communication and addressing.
The idea is that the fingerprint of the public key is used as the address of the device. Nobody can impersonate another person, because not having the matching private key, he or she won't be able to decrypt the message and negotiate the symmetric AES key used for further communication.
I have proposed a similar solution for WiFi mesh networks 8 years ago - https://groups.google.com/g/comp.os.linux.networking/c/4Cu_k2lYCSQ/m/WzlZ8BBHvSQJ , but never had time and/or knowledge sufficient to implement it.
If the described idea is a nonsense, please disregard my proposal.

a-f-G-U-C, 2020-09-30:
A great suggestion.

I'm sorry, the linked article was login-walled for me, but I'll comment on the concept in general.

The public-key-(hash)-as-identity concept has been used (naively) by several decentralized projects out there, usually with disappointing results. Off the top of my head here are some of the issues and pitfalls:

**1) No forward secrecy.** I want to be able to change my encryption key frequently, in order to minimize the impact of a key compromise on past and future communications. Linking my network identity to my encryption key prevents this, or at least makes it very inconvenient. 

**2) Metadata persistence.** I want to change my mesh node ID (address) frequently, in order to prevent the bad guys from tracking the activity of individual nodes or learning the social graph of my group. Linking the node ID to my key, or network identity, limits this ability.

**3) Group messaging.** (EC)DH-based key establishment protocols work great for point-to-point comms, not so great for point-to-multipoint. 

However, if these issues were handled properly - and there's a whole lot of information out there on how to do it right - we could be looking potentially at a **really useful** system (think OTR over Bitmessage over mesh radio - what's not to love).

Anyway, I suggest this discussion might work better in the Meshtastic-Android project as the typical Meshtastic-device hardware isn't really suitable for protocols based on brute-force EC cryptography.

wzab, 2020-09-30:
I have edited the link, so that it does not require login.
Indeed you've pointed out significant drawbacks of the original proposal.
I'd be glad if it can be an inspiration for a solution providing really private communication via Meshtastic. 
In fact ESP32 should offer sufficient computational power to support RSA with 4096-bit key.
Maybe I'm paranoid but I'd prefer to have everything that enters an Android device already encrypted...
That's why I'd like my Meshtastic device be equipped with independent UI to enter the message to be sent.

geeksville, 2020-10-05:
good ideas.

bsiege, 2020-10-17:
Encryption is important. 
But i do not see a classical Diffie Hellman Key exchanges for LORA. Less a problem from a computational view, but a problem of packet sizes and duty cycles in RF-Part. Way too much overhead. Encryption like implemented in LoraWAN could be a solution. 
But: If you prepare for disaster maybe you want a open transmission protocol to easy expand the mesh and then have the possibility to generate closed groups later simply with a shared key. 
So, "bullet-proof" privacy is IMHO less important in a case of "Disaster-Radio". If you need a "Underground-Resistance-Radio" this is quite a different story :-)

wzab, 2020-10-18:
> So, "bullet-proof" privacy is IMHO less important in a case of "Disaster-"Radio". If you need a "Underground-Resistance-Radio" this is quite a different story :-)

I think it's good to be prepared for both use cases. Just in case...