meshtastic/firmware #452: Implement HTTPS

mc-hamster, 2020-10-01:
Implement an HTTPS service.

mc-hamster, 2020-10-01:
I did a POC today and will utilize the esp32_https_server as it supports both HTTP and HTTPS in parallel.

https://github.com/fhessel/esp32_https_server

I'm out this weekend but will be able to migrate Meshtastic over to this over the course of next week.

slavino, 2020-10-01:
So the concept here is to have HTTPS (SSL) protected REST API on each Meshtastic device capable of WiFi/wired network connection?

mc-hamster, 2020-10-01:
You got it!

mc-hamster, 2020-10-03:
Benchmark results. Saving it here incase we can tune this later.

**HTTP**

Server Software:        
Server Hostname:        10.10.30.198
Server Port:            80

Document Path:          /asdf
Document Length:        169 bytes

Server Software:        
Server Hostname:        10.10.30.198
Server Port:            80

Document Path:          /
Document Length:        164 bytes

Concurrency Level:      1
Time taken for tests:   1.238 seconds
Complete requests:      10
Failed requests:        0
Total transferred:      2080 bytes
HTML transferred:       1640 bytes
Requests per second:    8.08 [#/sec] (mean)
Time per request:       123.786 [ms] (mean)
Time per request:       123.786 [ms] (mean, across all concurrent requests)
Transfer rate:          1.64 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        6   18  34.6      7     116
Processing:    27  106  29.3    118     119
Waiting:       22   74  20.9     84      86
Total:         95  124  11.5    125     143

Percentage of the requests served within a certain time (ms)
  50%    125
  66%    125
  75%    125
  80%    125
  90%    143
  95%    143
  98%    143
  99%    143
 100%    143 (longest request)

**HTTPS**

Server Software:        
Server Hostname:        10.10.30.198
Server Port:            443
SSL/TLS Protocol:       TLSv1.2,ECDHE-RSA-AES256-GCM-SHA384,2048,256
Server Temp Key:        ECDH P-384 384 bits

Document Path:          /
Document Length:        164 bytes

Concurrency Level:      1
Time taken for tests:   19.772 seconds
Complete requests:      10
Failed requests:        0
Total transferred:      2080 bytes
HTML transferred:       1640 bytes
Requests per second:    0.51 [#/sec] (mean)
Time per request:       1977.229 [ms] (mean)
Time per request:       1977.229 [ms] (mean, across all concurrent requests)
Transfer rate:          0.10 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:     1879 1939  77.5   1915    2152
Processing:    32   38   4.6     40      43
Waiting:       14   15   1.2     16      17
Total:       1921 1977  78.8   1953    2195

Percentage of the requests served within a certain time (ms)
  50%   1953
  66%   1968
  75%   1974
  80%   1985
  90%   2195
  95%   2195
  98%   2195
  99%   2195
 100%   2195 (longest request)

mc-hamster, 2020-10-17:
This is done.

closed 2020-10-17 by mc-hamster