meshtastic/firmware #455: Need to refactor?

nils-werner, 2020-10-03:
With the ever increasing selection of (potentially) compatible LoRa boards, I see an issue of exploding complexity in this project: Each board may potentially ship a different kind of LoRA, GPS, BT, screen, SoC etc. that need to be supported.

Right now, as far as I can tell, these different boards are supported via `#ifdef` clauses littered all over the codebase. In the future, this approach may at some point become too complex to maintain. Therefore I would like to start a conversation about refactoring this project to have some sort of "component drivers", like the Linux kernel does.

What do you think, is this something this project needs at some point?

geeksville, 2020-10-05:
Oh yes, that's something we definitely need.  Would you be interested in working on that?

nils-werner, 2020-10-06:
Yes, I'd like to help with that. Although I must say I have no experience with writing HALs, so I will have to learn how to do that while doing this :-)

nils-werner, 2020-10-07:
To start with this, I would say we should

 1. Find a terminology for the components in this project
 1. Find hardware-abstract tasks that describe what each component and device must be able to do (e.g. init, sleep, deep sleep, wake up, send data, draw pixels etc.)
 1. Find a neat dir/file structure for the individual component source files
 1. Find a way to configure the inclusion/exclusion of components and drivers for specific boards
 1. Define the API described in 2
 1. Document steps 1-5
 1. Refactor the current implementation in the newly designed structure

geeksville, 2020-10-08:
I've done a fair number of HALs for different products and I agree with that plan.  Though I'd add we should leverage C++ inheritance as much as possible when doing the cleanup.  If you are hip to work on this and want to send in a series of atomic commits that sounds great!

nils-werner, 2020-10-08:
I wasn't sure if C++ was supported in all important toolchains, but I guess it is.

Maybe it is too early to talk about this, but thinking more about C++ inheritance, I was wondering how we would solve this at a

mc-hamster, 2021-03-27:
closing the request for now. we had a major refactor as part of the 1.2 effort and the web server was refactored a few months ago.

closed 2021-03-27 by mc-hamster