meshtastic/firmware #461: LoraWAN in New Zealand

dh47, 2020-10-09:
From what I see the band plan to use is 915MHz.  But witch one do I use?  ANZ or US?  When I use the US I see the pings in the 915 range, but not when using ANZ.

Regards,

a-f-G-U-C, 2020-10-11:
AFAIK Meshtastic does not use LoRaWAN. It uses the LoRa P2P (non-WAN) protocol on custom channel plans, carefully designed to **avoid** rather than overlap those used by LoRaWAN.

geeksville, 2020-10-12:
@dh47 for NZ you would be best off using the ANZ setting (which was contributed by a New Zealander).  

(And github is fine for questions like this but in the future you would probably get a quicker/broader answer from our forum: meshtastic.discourse.group)

closed 2020-10-12 by geeksville