meshtastic/firmware #475: TTGO LoRa V1 - Bluetooth pairing forgotten after power-cycle

in03, 2020-10-12:
### Bluetooth pairing forgotten after power-cycle

**Steps to reproduce the behavior:**
1. Power on TTGO Lora V1 board running Meshtastic
2. In Meshtastic Android app, pair to Meshtastic device
3. Power off the TTGO Lora V1 board
4. Power on the TTGO Lora v1 board
3. Wait... After 15 minutes or so, the Android phone will request to be re-paired and the new pairing code will appear on the Meshtastic device. Every re-pair after this power-cycle is temporary and the devices will endlessly request to be paired at random intervals between 10 seconds and 15 minutes or so.
5. Go to Android phone's bluetooth settings and forget the Meshtastic device.
6. Repeat step 1 & 2. The phone will stay paired with no re-pairing issues, so long as the device isn't power-cycled.

**Expected behavior:**
After the initial pairing, the TTGO Lora V1 board running Meshtastic and the Anrdroid phone should not require to be re-paired unless either device is made to "forget" the pairing.  

**Device info:**
 - TTGO Lora V1
 - Software Version 1.1.0

**Smartphone info:**
 - Device: Samsung Galaxy A31 (SM-A315G/DS)
 - OS: Android 10
 - App Version 1.1.00

**Serial Output:**
[TTGO LoRa V1 - Forgotten bluetooth pair.txt](https://github.com/meshtastic/Meshtastic-device/files/5363928/TTGO.LoRa.V1.-.Forgotten.bluetooth.pair.txt)
[TTGO LoRa V1 - Healthy bluetooth pair.txt](https://github.com/meshtastic/Meshtastic-device/files/5363929/TTGO.LoRa.V1.-.Healthy.bluetooth.pair.txt)

**Guessing this is the problem?**
```failed to configure restored IRK```

geeksville, 2020-10-12:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/board-recommendations/1402/18

mc-hamster, 2021-03-27:
This described behavior no longer happens on our recent builds. 

Closing this request.

closed 2021-03-27 by mc-hamster