meshtastic/firmware #476: Coding style - use consistent integer formatting in debug messages

a-f-G-U-C, 2020-10-12:
## Background
The firmware debug messages use a combination of hex, unsigned decimal and signed decimal formats to represent integer values. 

## Currently
There are several problems:
1) many values are not printed in the best, or most readable format
2) sometimes the **same variable** can be printed using **different formats** in various parts of the code, making it harder to filter in the output logs
3) *accidental integer signedness* makes me **physically sad** :)

**Actual example:**
```
Rx someone rebroadcasting for us (id=0xfc882b32 Fr0xef To0xff, WantAck0, HopLim0 encrypted rxSNR=5.5)
Someone is retransmitting for us, generate implicit ack
Sending an ack=0x1,to=0xb224beef,idFrom=-58184910,id=858395619
[...]
Adding packet record (id=0x332a13e3 Fr0xef To0xef, WantAck0, HopLim0 Payload:None successId=fc882b32 rxtime=1602500000)
```
Conversion cheat sheet:
- 0xfc882b32 = **-58184910 (!)** = fc882b32
- 858395619 = 0x332a13e3

As seen above, having the same variables represented in 2.5 different formats makes the code awkward to debug, via either visual or machine inspection of logs. Imagine using `grep` on this log to follow the lifecycle of a problematic message.

## Proposal
It would be really helpful if the project could adopt and apply a consistent formatting style, based on the formats that make the debug output most readable, actionable and machine-parseable. While I don't have a recommendation for any pre-made set of rules, here are some rules of thumb that I personally use:
- if an **industry standard** representation exists (like CIDR IPv4 etc), **use it**; otherwise:
- use **decimal** for quantities and positions, and mind the signedness
- use **hex** for numeric identifiers, bignumbers (keys/hashes) and binary data, zero-padded to namespace size and prefixed by 0x

Thank you.

mc-hamster, 2020-10-12:
Just a data point ... From the example, "Sending an ack" is a debug message with conscious formatting:

DEBUG_MSG("Sending an ack=0x%x,to=0x%x,idFrom=%d,id=%d\n", isAck, to, idFrom, p->id);

while  "Adding packet record" is printed as a serialization of MeshPacket *p:

printPacket("Adding packet record", p);

geeksville, 2020-10-13:
I think this is a great proposal.

geeksville, 2020-12-13:
I think I've just fixed this mostly by using printPacket() more uniformly.

closed 2020-12-13 by geeksville